RIOT-OS/RIOT #8130: gcoap: can't build with network stacks other than GNRC

miri64, 2017-11-23:
If one tries to build gcoap with any other stack than GNRC (e.g. by adding gcoap to the `tests/lwip` application), it won't build. This is because `gcoap` uses the (always stack dependent) member `reg` of the `sock_udp_t` type are accessed in `gcoap_req_send2()`.

gebart, 2017-11-23:
Doesn't the G in gcoap stand for GNRC?

miri64, 2017-11-23:
As far as I know that has legacy reasons. Also, then why is it using `sock`? Also also: this one line is the only thing preventing `gcoap` to be used with other network stacks ;-)

kb2ma, 2017-11-24:
Originally, gcoap was developed for GNRC. However, conn/sock was implemented in the same timeframe, so I adapted gcoap to sock. As Martine has found, the adaptation is not complete. I should have dealt with this long ago. :-/

gcoap runs in its own thread, and listens indefinitely as a server in sock_udp_recv(). However, when another thread wants to send a message, gcoap puts a message in the sock recv mbox to interrupt that listening. After the send, gcoap uses timed listening while waiting for the message response. I have documented this better in #7337 [1]. See the mbox_try_put() in gcoap_req_send2() and sock_udp_recv() in _listen().

Taking a quick look at lwip, it looks like I can use the same approach using the netconn.recvmbox. So then I can use a conditional macro based on the network stack being used (LWIP, GNRC) to choose the right struct member for the mbox for the message.

As I understand it, the sock API was meant for application-level use, but gcoap operates a little deeper. Hopefully, it will take just a small violation of the abstraction to get it to work here. It's a long holiday weekend here, so I'll look into this next week. Martine, I'd appreciate your thoughts on the approach. Thanks!

[1] https://github.com/kb2ma/RIOT/blob/3de19fdd246338499e8336ddda6614be9b2e7793/sys/net/application_layer/gcoap/gcoap.c

miri64, 2017-11-24:
> Taking a quick look at lwip, it looks like I can use the same approach using the netconn.recvmbox. So then I can use a conditional macro based on the network stack being used (LWIP, GNRC) to choose the right struct member for the mbox for the message.

I think that just stashes the problem a step further down ;-).

I think what we really need so your gcoap thread does not have to listen indefinitely is to finally add some events to `sock` (so we have something like `select()`). Now that @kaspar030 event abstraction #6000 is merged this should be pretty simple to integrate. As soon as I'm done with #7925 I will start to think about that.

kb2ma, 2017-11-24:
Thanks, sounds like a good plan.

miri64, 2018-02-14:
BTW the PR for that is already open for a while: #8149.

kb2ma, 2018-03-05:
I'm experimenting with #8149 / #8236 in gcoap. So far, so good. I expect that this work will provide a simpler interface for waiting than the mbox and timer used currently. I hope to have something to share later this week.

miri64, 2019-12-07:
@kb2ma did you try again with #12625 and #12602

kb2ma, 2020-01-13:
I'm stumbling on integration of sock events in #12602. How do I set a timer on a request [here](https://github.com/RIOT-OS/RIOT/blob/76cfe2b84c8dfc8d2f19140981b3907e313781b6/sys/net/application_layer/gcoap/gcoap.c#L803) and handle it?

Let's assume I have an event queue for the UDP (or DTLS) sock. At the location linked above, what should be done instead of the mbox_try_put()? Can I just remove that function call? How do I designate the event for the timeout -- still with xtimer_set_msg()? Or do I somehow need to generate a timeout event in the sock event queue?

miri64, 2020-01-13:
Doesn't [`event_timeout`](https://github.com/RIOT-OS/RIOT/blob/master/sys/include/event/timeout.h) work for that? If not we maybe can maybe extent that event type for your needs.

kb2ma, 2020-01-14:
Thanks for the pointer; I think I get it. Looks like I'll need to create a gcoap event for a response timeout.

It will be nice to have a single mechanism for sock events and coap events. It's a significant change though.

miri64, 2020-01-14:
> […] It's a significant change though.

That was to be expected.

kb2ma, 2020-02-13:
As a stopgap, anyone interested in gcoap working on lwIP and IPv4 please see my [gcoap/lwip_for_example](https://github.com/kb2ma/RIOT/commits/gcoap/lwip_for_example) branch.

miri64, 2020-03-06:
Fixed with #13386 and tested during the testing for #13427. So let's close this finally :-)

closed 2020-03-06 by miri64