RIOT-OS/RIOT #8133: gnrc_rpl: missed edge cases with route lifetimes

bergzand, 2017-11-24:
2 edge cases were missed with #8128, both for the preferred lifetime and for the valid lifetime:

1. Prefixes with infinite lifetimes should be advertised as infinite. Currently when a prefix has an infinite lifetime on the RIOT node (`UINT32_MAX`), it is divided by `MS_PER_SEC` and added to the packet. This should be added as `UINT32_MAX` instead, as that is the indication for RPL information that the prefix lifetime is infinite. Same on the receiving (parser), lifetimes received as `UINT32_MAX` should not be multiplied with `MS_PER_SEC`, but directly added as `UINT32_MAX`.
2. Prefixes received with lifetime larger than `(UINT32_MAX - 1)/1000` should truncated to as long as possible without causing it to be infinite. This problem is caused by RPL using lifetime in seconds and RIOT using lifetime in milliseconds and both being stored in 32 bits. The RPL prefix lifetime option is thus able to handle a larger range of lifetimes

bergzand, 2017-11-24:
For case number 2 it might be enough to add those routes as `UINT32_MAX - 1`

miri64, 2017-11-24:
Or I do it as I did with the PIO in the RAs ;-)

miri64, 2017-11-24:
I do not handle corner case 2 there correctly as well... will fix as well.

miri64, 2017-11-24:
Fixed in #8135.

bergzand, 2017-11-24:
#8135 is merged

closed 2017-11-24 by bergzand