RIOT-OS/RIOT #8141: Fix common python style issues found in automatic test scripts

aabadie, 2017-11-27:
Here is the list of the problems returned (only in `tests`):
- [ ] E402 module level import not at top of file (because of `import testrunner` after `sys.path.append`...)
- [ ] E302 expected 2 blank lines, found 1
- [ ] F401 *module* imported but unused
- [ ] E401 multiple imports on one line
- [ ] E128/E127 continuation line under/over-indented for visual indent
- [ ] E711 comparison to None should be 'if cond is not None
- [ ] E502 the backslash is redundant between bracket
- [ ] E241 multiple spaces after ':'
- [ ] E202 whitespace before ')'
- [ ] E231 missing whitespace after ','
- [ ] E501 line too long (270 > 119 characters)
- [ ] E225 missing whitespace around operator
- [ ] E226 missing whitespace around arithmetic operator
- [ ] E123 closing bracket does not matc indentation of opening bracket's line
- [ ] E703 statement ends with a semicolon (this is not C code !!)
- [ ] F841 local variable *variable* is assined to but never used

I'll come up with PRs fixing each of them to simplify reviews

aabadie, 2017-11-27:
I would also like to have opinions on 'E402 module level import not at top of file'. There are several ways to remove this:
* move the `import testrunner` in the `__main__`. Already done in other scripts but I'm not sure that's the cleanest
* in the test Makefile, export an updated `PYTHONPATH` environment variable containing the path to `testrunner.py` just before the call to the script. Then no need to use the `sys.path.append` in the script. I tested it and it also works.
* make a package of `testrunner` and install it system/user wide

cladmi, 2017-11-28:
We discussed with @aabadie regarding python conventions.

He will openan issue to propose a documentation.

In short, `python3.5` compatible (because its ubuntu stable version) and following python reference style https://www.python.org/dev/peps/pep-0008/

kaspar030, 2017-11-29:
> in the test Makefile, export an updated PYTHONPATH environment variable containing the path to testrunner.py just before the call to the script.

I'd prefer this. Should be a one-liner in the makefiles, right?

aabadie, 2017-11-29:
> I'd prefer this. Should be a one-liner in the makefiles, right?

yes. After discussing IRL with @cladmi, we found that using the first solution (moving sys.path.append and import testrunner in the `__main__` block was the better solution. Then the test script will just be:
1. import code module `os`, `sys` and necessary modules for the test
2. (optional) define global variables
3. write a testfunction
4. put the `__main__` block containing the import to testrunner and the call to `run`.

See [here](https://github.com/RIOT-OS/RIOT/blob/master/tests/pkg_u8g2/tests/01-run.py) for an example.

aabadie, 2017-11-29:
> He will open an issue to propose a documentation.

Finally, I'll post a documentation draft in markdown here. This draft, if accepted, can be later use as wiki page for Python script convention.

Draft:

## Python coding convention

### General

* Code shall be compliant with Python 3.5 minimum because this is the default Python 3 version in Ubuntu 16.04 (used as the reference system for CI)
* Code shall be [PEP8](https://www.python.org/dev/peps/pep-0008/) compliant
* A runnable script shall start with `#!/usr/bin/env python3`
* A line length of maximum 120 characters is allowed: this is because the testrunner scripts can expect potentially long output lines from RIOT test applications

### Testrunner scripts

* `testrunner` scripts are placed in a `tests` subfolder of a test application
* `testrunner` scripts names follow the pattern `<number>-run.py`
* `testrunner` scripts have the following structure:
```python
#!/usr/bin/env python3

# Copyright (C) <your copyright>
#
# This file is subject to the terms and conditions of the GNU Lesser
# General Public License v2.1. See the file LICENSE in the top level
# directory for more details.

# Global imports
import os
import sys

# Test specific imports
import module1
import module2

# Optional global variables
GLOBAL_VARIABLE = "I'm global"

def testfunc(child):
    # Put your test code here

if __name__ == "__main__":
    # import testrunner script from RIOT tools
    sys.path.append(os.path.join(os.environ['RIOTBASE'], 'dist/tools/testrunner'))
    from testrunner import run
    sys.exit(run(testfunc))

```

aabadie, 2017-11-29:
@cladmi, here is a draft for a Python script coding convention page, feel free to amend the comment.

miri64, 2017-11-29:
> * `testrunner` scripts names follow the pattern `<number>-run.py`

Why this restrictive naming convention? I would find it actually pretty useful, that if there are multiple test scripts (for different test cases e.g.) that they could have a more decriptive name, like

* `01-send-udp.py`
* `02-icmpv6-ping.py`

aabadie, 2017-12-01:
> Why this restrictive naming convention?

I have no particular opinion on that. Since the script are called from the makefile we can choose any name. Here I just followed what is currently used in the test.
Your suggestion makes sense to me because it's self explanatory and keep some kind of name ordering. I can change that "rule"

miri64, 2017-12-03:
> I have no particular opinion on that. Since the script are called from the makefile we can choose any name. Here I just followed what is currently used in the test.
> Your suggestion makes sense to me because it's self explanatory and keep some kind of name ordering. I can change that "rule".

I think *I* started that, because the initial `pexpect` script by someone else was called like that.. But I never saw it as a general rule, more like lazy copy-pasta. IMHO (no technical, no taste-wise at least) there is no reason to put a constraint on the name so we shouldn't. For the Makefiles we could just use `$(shell find tests/ -x -type f)` to find all executables in `tests/`.

aabadie, 2017-12-04:
@miri64, I think it's important to have minimal rules on the test scripts and probably having something like `<number>-<name>.py` is good. The idea is just to guide new contributors and avoid strange/meaningless names on test scripts.

@cladmi do you have comments on the proposed [Python coding conventions](https://github.com/RIOT-OS/RIOT/issues/8141#issuecomment-347912851) ?

miri64, 2017-12-04:
> The idea is just to guide new contributors and avoid strange/meaningless names on test scripts.

I'd argue that `01-run.py` already is pretty meaningless, but I agree, that the pattern `<number>.*<name>` for scripts that are run in batch is a time-proven naming scheme ;-).

miri64, 2017-12-04:
> testrunner scripts have the following structure:

Next gripe to pick: If we ever change how the testrunner works (which from what I've heard might be pretty soon), so why enforce this structure? Not even now every test follows this structure.

aabadie, 2017-12-04:
> Not even now every test follows this structure.

Not yet, but it's a matter of time if we fix remaining pep8 and especially the E402 one (because the `sys.append` is called before `import testrunner`)

aabadie, 2017-12-04:
> If we ever change how the testrunner works

Indeed but this is not the case **now**. If it changes one day, we'll just have to update this convention.

miri64, 2017-12-04:
> Indeed but this is not the case now. If it changes one day, we'll just have to update this convention.

My problem is not that the convention needs to change (though I think it is weird a *convention* must be changed once the framework changes, which should already ring some alarm bells), but that a rigid structure is a supposed to be a convention.

miri64, 2017-12-04:
> The question could also be: should we enforce the use of testrunner for simple automatic python test scripts?

No, because it is very restricted to a certain kind of test.

aabadie, 2017-12-04:
> certain kind of test

which ones ? do you mean network tests ?

miri64, 2017-12-04:
Network tests, HIL tests, any test that involves more than one board or even more than one process (i.e. child in the context of `testrunner`) to control using `pexpect`.

aabadie, 2017-12-04:
Ok, we are thinking the same. So to keep the test scripts more generic, we can just give a generic skeleton script without references to trestrunner related stuff  and put a testrunner specific template structure. What do you think ?

miri64, 2017-12-04:
I'm not against providing a skeleton script. We do [this already for application Makefiles](https://github.com/RIOT-OS/RIOT/blob/master/dist/Makefile). I'm against (hard NACK in fact) writing it into the coding convention for python scripts.

aabadie, 2017-12-04:
> I'm against (hard NACK in fact) writing it into the coding convention for python scripts.

I don't understand why. The generic skeleton is just a common rule for a regular python script (with `if __name__ == '__main__': etc. The rules given above is just for testrunner script and to me it makes sense to have it.

miri64, 2017-12-04:
Because I don't see any reason to enforce something like that. If you know any, except for being able to bug contributors about this ;-), please name some.

miri64, 2017-12-04:
The `if __name__ == "__main__":` thing is part of python conventions (because there is a technical reason for it; if it isn't in there you can not load the script as a module without executing the code everytime). But this is about *one* line in the entire script. Your suggestion gives a mandatory order, a mandatory naming and a mandatory call structure to the script that doesn't really sit right with me, because it is way too restrictive and bound to a specific framework that is slated to be replaced in the near future anyway.

aabadie, 2017-12-04:
> Your suggestion gives a mandatory order, a mandatory naming and a mandatory call structure to the script that doesn't really sit right with me, because it is way too restrictive and bound to a specific framework that is slated to be replaced in the near future anyway.

It seems you are discussing on implementation details here and maybe the testrunner improvement (#7258) is introducing some kind of bias here.
I propose to simply remove any specific rules related to testrunner and keep general rules on how a runnable Python test script should look like. Is it ok ?

I haven't heard about any decision for that, only discussions. The actual simple testrunner scripts are ok to me and shouldn't be changed.

miri64, 2017-12-05:
> It seems you are discussing on implementation details here and maybe the testrunner improvement (#7258) is introducing some kind of bias here.

No bias, but there is IMHO a lesson to be learned that frameworks are not necessarily here to stay.

> I propose to simply remove any specific rules related to testrunner and keep general rules on how a runnable Python test script should look like. Is it ok ?

I think PEP8 and other common Python conventions should suffice or do you have more in mind?

aabadie, 2017-12-11:
@cladmi, @miri64, let's move forward here. What do you think of the actual python [coding conventions above](https://github.com/RIOT-OS/RIOT/issues/8141#issuecomment-347912851) ?

miri64, 2017-12-11:
As they are I'm okay with them.

aabadie, 2017-12-11:
> As they are I'm okay with them.

Yeah :)

cladmi, 2017-12-11:
* Specifying the python version is good, also the shebang for executable scripts.
* Imposing the `if __name__ == '__main__':` is necessary to also allows some checkers to correctly execute, because in other cases it could execute the 'main' function as part of dynamic parsing. I already had this and its bad.
* uppercase global variables naming is also in PEP8.
* The order of imports is defined in PEP8 https://www.python.org/dev/peps/pep-0008/#imports and checked, by `pylint` for example so yes

So OK for all of this.

Points I would like to adapt, mainly remove 'tests' mentions the rest is just nitpicking.

* I would like to remove all the `tests` specific mentions as its a general python convention (only needed in the 120chars limit description).
* Use PEP8 text for order of imports, not rephrasing our own version (or just link to it)
* Rephrase the 120 chars limit: " ... of 120 is allowed instead of the pep8 79: this increases tests readability as they can expects long line of output.""

aabadie, 2017-12-11:
@cladmi, I think I addressed your comment. Not sure about the order of imports but I added a link to the corresponding pep8 section in the script example.

cladmi, 2017-12-12:
That's good, last fix is two spaces before `if __name__ == '__main__'` according to running `flake8`.

aabadie, 2017-12-12:
AFAICT, there's no explicit rules about whether one or two blank lines are required before `if __name__ == __main__:` see this [SO answer](https://stackoverflow.com/a/15889078) and the [blank lines section](https://www.python.org/dev/peps/pep-0008/#blank-lines) in pep8.

I would leave the actual convention as it is.

cladmi, 2017-12-12:
I just added 'pass' to the function definition to make it work and get this output with flake8.

```
flake8 rules.py
rules.py:12:1: F401 'module1' imported but unused
rules.py:13:1: F401 'module2' imported but unused
rules.py:30:1: E305 expected 2 blank lines after class or function definition, found 1
```

cladmi, 2017-12-12:
```
flake8 --version
3.5.0 (mccabe: 0.6.1, pycodestyle: 2.3.1, pyflakes: 1.6.0) CPython 3.5.2 on Linux
```

aabadie, 2017-12-12:
Ok, in fact pep8 says that the top level def or class should be _surrounded_ with 2 blank lines, so you are right.

aabadie, 2017-12-12:
Convention updated !

cladmi, 2017-12-12:
ACK

aabadie, 2017-12-12:
> ACK

Section added to coding convention wiki page: https://github.com/RIOT-OS/RIOT/wiki/Coding-conventions#python-coding-convention

smlng, 2017-12-15:
I think this is through and can be close, if you @aabadie agree: please hit the close button.

aabadie, 2017-12-15:
> I think this is through and can be close

Indeed, let's close !

closed 2017-12-15 by aabadie