RIOT-OS/RIOT #8151: style: notation of hexadecimal numbers (0xAFFE vs 0xaffe)

haukepetersen, 2017-11-27:
While reviewing #7937, I noticed that we don't have a preferred/default/defined way to note hexadecimal numbers: in the RIOT code base they are as often written using capital letters as they are using lowercase. 

I think it would make sense to add a sentence with a preferred style to the coding conventions. Personally I prefer lowercase letters, but this is certainly very subjective. So how about a quick vote: (i) "uppercase", (ii) "lowercase", (iii) "absolutely indifferent and this is a waste of time".

aabadie, 2017-11-27:
Why not use Github emoji for the vote ? Let's add a comment for each proposition and people vote the comment they prefer. What do you think ?

miri64, 2017-11-27:
I prefer (ii). This is personal taste (and a weird one as well), but uppercase hex look always kind of MS-DOS-y and old-timey to me.

haukepetersen, 2017-11-27:
@aabadie: alright, we might as well try it. So please vote on one of the tree following:

haukepetersen, 2017-11-27:
use uppercase for hexadecimal numbers: `0xAFFE`

haukepetersen, 2017-11-27:
use lowercase for hexadecimal numbers: `0xaffe`

haukepetersen, 2017-11-27:
I am indifferent. And this is a waste of time. And I really really don't care.

aabadie, 2017-11-27:
Please vote only once !

haukepetersen, 2017-11-28:
Seems like the majority of people doesn't care, so I take we simply do nothing w.r.t. this topic. Closing this issue again - as always: re-open of anyone disagrees with the outcome.

closed 2017-11-28 by haukepetersen