RIOT-OS/RIOT #8162: MODULE_CCN_LITE error

hahchenchen, 2017-11-28:
I add `USEMODULE += ccn_lite` in Makefile of examples/ccn-lite-relay. my board is nucleo144-f429

there came out en error when compile:
`arm-none-eabi-gcc: error: /home/c/ICN/RIOT/examples/ccn-lite-relay/bin/nucleo144-f429/ccn_lite.a: No such file or directory`

is there any help for that?

PeterKietzmann, 2017-11-28:
@hahchenchen it is `USEPKG += ccn-lite`. However, there are a couple of other requirements. Did you loog at the ccn-lite-relay example, especially the [`Makefile`](https://github.com/RIOT-OS/RIOT/blob/master/examples/ccn-lite-relay/Makefile)?

miri64, 2017-11-28:
For reference (and maybe more discussion) on this https://github.com/RIOT-OS/RIOT/issues/6668.

hahchenchen, 2017-11-28:
Yes, this my setting:
`USEPKG += ccn-lite`
`USEMODULE += ccn-lite-utils`
I wanna add`USEMODULE += ccn_lite`, because in the nettype.h: https://github.com/RIOT-OS/RIOT/blob/master/sys/include/net/gnrc/nettype.h#L101, i need ccn_lite module for enable the GNRC_NETTYPE_CCN protocol.

hahchenchen, 2017-11-28:
Hm,  my network device is cc110x, so i need add a case for GNRC_NETTYPE_CCN in _send function: https://github.com/RIOT-OS/RIOT/blob/master/drivers/cc110x/gnrc_cc110x/gnrc_cc110x.c#L64
right?

PeterKietzmann, 2017-11-28:
@hahchenchen sorry for not reading properly. Don't use `USEMODULE += ccn_lite`. The nettypes are already included with that example, because `MODULE_CCN_LITE` is automatically generated. Yes, is seems the driver needs adaption. See what was done for the [Amel 802.15.4 driver](https://github.com/RIOT-OS/RIOT/blob/master/drivers/at86rf2xx/at86rf2xx.c#L92).

PeterKietzmann, 2017-11-28:
YOu referenced the `_send` function. In the [`_recv`](https://github.com/RIOT-OS/RIOT/blob/master/drivers/cc110x/gnrc_cc110x/gnrc_cc110x.c#L131) function it seems to be handled properly. Please just remove `USEMODULE += ccn_lite` from your makefile and report the actual issue you have

hahchenchen, 2017-11-28:
Hm, the actual issue is: when i use cc110x as a network device to run ccn-lite, 
the gnrc_netreg_num function always return 0: https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/netreg/gnrc_netreg.c#L83

PeterKietzmann, 2017-11-28:
Did you include the radio driver to your Makefile  (`USEMODULE=cc110x`)? When I do so I get the following error:
```
RIOT/sys/auto_init/netif/auto_init_cc110x.c:30:27: fatal error: cc110x_params.h: No such file or directory
```
... which is due to an inconsistency with that driver. The msba2 board inculdes such a fiel, but there is no fallback (section "Default device configuration" in [this guide](https://github.com/RIOT-OS/RIOT/wiki/Guide:-Writing-a-device-driver-in-RIOT)) might be of interest. How did you connect the radio and which configuration did you use?

hahchenchen, 2017-11-28:
I include the `USEMODULE=cc110x`, and i modify cc110x_params.h in msba2 to suit my board.

miri64, 2017-11-28:
Did you include `USEMODULE += auto_init_gnrc_netif`

hahchenchen, 2017-11-28:
Yes

hahchenchen, 2017-11-28:
I found that when cc110x recv the pkg, and called the gnrc_netreg_num funciton, the type is GNRC_NETTYPE_UNDEF, but the entry for that is `NULL`:https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/netreg/gnrc_netreg.c#L94
 It seems we don't register netreg for GNRC_NETTYPE_UNDEF

miri64, 2017-11-28:
>  found that when cc110x recv the pkg, and called the gnrc_netreg_num funciton, the type is GNRC_NETTYPE_UNDEF, but the entry for that is NULL:https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/netreg/gnrc_netreg.c#L94
It seems we don't register netreg for GNRC_NETTYPE_UNDEF

Ah yes of course. You need to search for `GNRC_NETTYPE_CCN`. Maybe some fix to the GNRC adapter of the `cc110x` is in order here.

hahchenchen, 2017-11-28:
Yes, i think i will provide a PR as soon as possible

PeterKietzmann, 2018-06-26:
@hahchenchen I assume the PR you've been talking about was #8191? We closed that one due to unresponsiveness. Furthermore, since you opened this issue there were a [couple of efforts](https://github.com/RIOT-OS/RIOT/pulls?q=is%3Apr+cc110x+is%3Aclosed) concerning the cc110x radio. Are you still willing to provide a fix?

PeterKietzmann, 2019-02-21:
For the protocol: #10340 provides a complete rewrite of the radio driver. I'm closing this issue due to lack of responsiveness.

closed 2019-02-21 by PeterKietzmann