RIOT-OS/RIOT #8184: Tests: periph_uart: HIL test required for development and CI testing

danpetry, 2017-11-30:
Tests: periph_uart: HIL test required for development and CI testing

# Summary

No automated testing currently exists for UART drivers. Such a test would aid driver development by providing a series of tests to run during development, both increasing speed of development and ensuring functional correctness. For the future, the test could be run by Murdock on permanent hardware setups, to avoid regression bugs. This would support current development of Murdock which includes the ability to run tests on hardware.

This discussion could potentially form a basis for some kind of standardization between tests to make integration into Murdock easier later on.

# Initial concept

Since UART0 is used for the RIOT shell, two ideas are given below; one for testing of UART0, and one for testing other UARTS.

Note that the idea for testing UART 0 can be used for testing the other UARTS as well.

## Testing UART0

[image]

- Tester runs periph_uart test application (which gives shell commands for setting up UARTs and sending data, directly interfacing with the UART API), with a known working version of the UART driver
- Board under test runs a UART echo application on the layer directly above the UART driver under development or test, which also changes the driver baudrate if it received a command with some simple format (e.g. b[baudrate])
- pexpect based python script delivers shell commands and reads output to/from the Tester board. This would send strings over the UART using the periph_test application and expect an echoed reply. It would also cycle through baudrates.
- Tests defined in testrunner using format described in https://github.com/RIOT-OS/RIOT/pull/7258

## Testing UART1 and above

[image]

- Tester runs periph_uart test application (which gives shell commands for setting up UARTs and sending data, directly interfacing with the UART API), with a known working version of the UART driver
- Board under test runs periph_uart test application with the UART driver under development or test
- pexpect based python script delivers shell commands and reads output to/from both the Tester and Board Under Test. Strings would be sent over the UART to and from both boards, with replies expected on the receiving board. It would also cycle through baudrates on both, via the shell commands in the periph_uart application
- Tests defined in testrunner using format described in https://github.com/RIOT-OS/RIOT/pull/7258

## Test data sent across UART

The messages sent over UART to the Board Under Test would be as follows

- Single bytes
- Sensible sized blocks of bytes
- Blocks of bytes larger than sizeof(size_t) (largest buffer that can be passed to uart_write())
- Series of 0xFFs
- Series of 0x00s
- Blocks of bytes at the wrong baudrate

# Points for discussion

- https://github.com/RIOT-OS/RIOT/issues/7300 describes an output format of [SUCCESS] or [FAILURE] from the test application running on the board. This is different from the above. I was going to implement the above because it's quicker (no need to change the test application), then we could standardise the test interface later. Or better to do the work up front?
- Is the periph_uart test adequate to run on the tester board, or would some lower-level type of bit-bashing be more suitable using GPIO pins? (eg this would allow a test which delivers noise over the UART). This would be similar to what @haukepetersen has mentioned for an I2C test - could aim for one standard UART/SPI/I2C/etc test application?
- The test outlined in "Testing UART0" could also be used to test UARTs 1 and above. Should it be?
- Any additions/removals to the data that should be sent?

bergzand, 2017-11-30:
First thing that comes to mind with your test setup: Would it be useful to attach a logic analyzer to the test setup. This way you don't have to rely on the interpretation of either of the boards in case of verification issues.

danpetry, 2017-11-30:
Thanks Koen. Were you thinking that could be part of the CI setup or for
on-desk development only?

Is there a need to discern between the two here?

bergzand, 2017-11-30:
My main reason for attaching an analyzer would be for the following: The UART is implemented in hardware, as far as I know, these hardware modules only return either the data and/or maybe some error bit in case of failure. I think this doesn't provide enough feedback when for example the baud rate is configured incorrectly.

To answer your question, I don't think it is really necessary for the CI, nice to have though, but not a hard requirement. For development I think it can save some valuable debugging time.

danpetry, 2017-11-30:
Ok very good point thanks. I'll include this in any information re on desk development. I'll clarify what is required for CI testing and what is required for on-desk testing where relevant.

danpetry, 2018-01-16:
@smlng @kaspar030 do you have any thoughts on the above? It would be good to get your input on how useful/appropriate this is for Murdock integration

MrKevinWeiss, 2018-12-03:
Can we close this issue since we have something like that now?  (see nightly HiL)

danpetry, 2018-12-03:
yep

closed 2018-12-03 by danpetry