RIOT-OS/RIOT #8204: pkg: port LittleFS

miri64, 2017-12-04:
Offline @benpicco told me very excited how much faster and smaller than SPIFFS [LittleFS](https://github.com/geky/littlefs) is. A quick glance at the project's README let's me think that this might be a nice file system to support for RIOT. Anyone wants to port it to RIOT?

gebart, 2017-12-04:
Certainly an interesting project. It seems quite new based on the commit history. Is there anyone in the community who is involved in the development of that project?

miri64, 2017-12-04:
> Is there anyone in the community who is involved in the development of that project?

Seems to be only one [contributor so far](https://github.com/geky/littlefs/graphs/contributors). @benpicco used to be quite active in the community (he used to be a student at FU Berlin), but apart from being a user for his work he doesn't seem to be involved in the project directly.

closed 2018-01-09 by smlng

gebart, 2018-01-13:
LittleFS was featured on Phoronix today: https://www.phoronix.com/scan.php?page=news_item&px=LittleFS-ARM-Embedded-FS

PR #8239 (2018-01-09): fs: add littlefs as a package
Just to avoid duplicate work, I'm pushing this. I started to work on littlefs package integration. I wanted to test it and see the performances.
I still have some issues and I don't know when I'll have time to continue working on it.

Fix #8204