RIOT-OS/RIOT #8211: tests: fail due to slow serial (re)connect after reset on macOS

smlng, 2017-12-05:
On macOS many automated tests script fail due to slow (re)connect of serial after board reset when running `make test`. Specifically all test scripts (e.g., `tests/thread_flood`) that match/expect the very first output, will fail because this output is never shown due to slow serial reconnect. The configurable start delay proposed in #8053 doesn't help here. 

A possible solution would be to delay boot of RIOT right in `core` during board or cpu init, which could be optional for test purposes only. Another work around is to add xtimer_sleep(5) as first statement in `main()` before any other output or function is called - but that is very inconvenient as it requires to adapt test applications.

miri64, 2017-12-11:
Can we step back a second and discuss if it makes sense to change *kernel behavior* for testing depending on what host platform it is compiled?

cladmi, 2017-12-11:
It should not be depending on host platform but on setting a configuration.

The goal is allowing anyone to run tests on his boards from his own computer with a not too complex solution.

Some persons, have the problem that they always lose the first seconds of output after their node resets.
As discussed in #8053 comments, smlng problem is not board specific, and maybe not even OS specific but only specific to his own setup, OSX + a usb hub.
This means a solution should only be set by explicitly enabling it and absolutely not by default.

Then, the reason to modify the *kernel* is that it should affect all `main` functions and so easier to modify what calls `main` than all tests main functions.

In practice I would do it by adding a by default disabled part of code, that when enabled, does an active wait before calling `main`.

But I am open to other proposals, that's why I asked to write an issue for this :)

kaspar030, 2018-04-09:
Any solution should not require to change the image-under-test. We might want to catch the very first printouts, or test uart output without uart input.

cladmi, 2018-04-09:
I agree it would be better.

If not possible, if its really that you cannot capture the first 5 seconds after a reset, I still prefer to test with a modified version than not testing at all :) I thought we were in that case, that's why I thought about adding a delay before the main function.
But according to https://github.com/RIOT-OS/RIOT/issues/8901 and @kYc0o it can be something else so we will try to find a correct solution.

kaspar030, 2018-04-09:
> If not possible, if its really that you cannot capture the first 5 seconds after a reset, I still prefer to test with a modified version than not testing at all :)

Yeah, if the tester doesn't work, fix the testee, not the tester! ;)

smlng, 2019-09-09:
re-tested with some boards on macOS, issues does not pop up anymore. Hence, closing

closed 2019-09-09 by smlng