RIOT-OS/RIOT #8212: Fix common issues with shell scripts

bergzand, 2017-12-06:
shellcheck is an application to statically check shell scripts for common errors. Most of these consists of small issues with the shell scripts where the problem might only appear in certain edge cases.

As an example, one of the issues is that shellcheck warns for is not having quotes around `${VAR}`. This issue leads in the headerguards check to vague errors as soon as there are spaces in `$PWD`.

I think this could also be a useful addition to the CI to check the shell scripts for common mistakes.

[1]: https://github.com/koalaman/shellcheck

gebart, 2017-12-06:
I like, more static tests 👍

aabadie, 2017-12-06:
Indeed, very nice. Would be great to have something similar for Makefiles ;)

Lotterleben, 2018-09-19:
Since both PRs are merged, can this issue be closed?

PeterKietzmann, 2018-09-25:
It seems so. @bergzand please reopen if I'm mistaken

closed 2018-09-25 by PeterKietzmann