RIOT-OS/RIOT #8213: at86rf2xx: Basic mode and NETOPT_AUTOACK

roberthartung, 2017-12-06:
1. Why is it not possible to choose the default RX_MODE (without AACK enabled)?

2. In at86rf2xx_netdev.c:_set(NETOPT_AUTOACK) the comment says `/* don't set res to set netdev_ieee802154_t::flags */`. However `netdev_ieee802154_set` never handles this cases which causes `ifconfig <n> set -autoack` to fail with an error message. What was the idea behind it?

roberthartung, 2017-12-06:
@haukepetersen @miri64 You should be able to answer these questions ;)

miri64, 2017-12-06:
> @haukepetersen @miri64 You should be able to answer these questions ;)

I don't know much about this device driver, except how it interacts with the `netdev` API ;-), but I can try to answer it with my limited knowledge:

> 1. Why is it not possible to choose the default RX_MODE (without AACK enabled)?

I could name you some villages in Bohemia, if you like ;-).

> 2. In at86rf2xx_netdev.c:_set(NETOPT_AUTOACK) the comment says /* don't set res to set netdev_ieee802154_t::flags */. However netdev_ieee802154_set never handles this cases which causes ifconfig <n> set -autoack to fail with an error message. What was the idea behind it?

Why would `netdev_ieee802154_set` handle that case? As far as I know auto-ack is pretty device specific, so this should be done in `at86rf2xx_netdev.c:_set()`.

roberthartung, 2017-12-06:
2) See #8216

PeterKietzmann, 2019-02-21:
@roberthartung  #8213 was merged shortly after you opened this issue. Could you give a quick summary of the remaining problem in the current state? If I see it correctly, point 1. was not addressed but I don't quite understand the problem there.

roberthartung, 2019-03-05:
@PeterKietzmann You mean #8216? ;-) 2. should be solved from my initial question, but 1. is still not possible, we are still always working in basic operation mode.

PeterKietzmann, 2019-03-08:
>  we are still always working in basic operation mode.

This confuses me a bit. I was assuming we only provide the extended mode in RIOT?

roberthartung, 2019-03-08:
You're absolutely right ... basic mode is not implemented. AACK is extended.

Btw. @PeterKietzmann do you mind if we implement the alternative frame buffer access mode? RIOT is currently writing the buffer first and then starting transmission which is slow. There is also a mode where you can write while preparing the send which is faster.

PeterKietzmann, 2019-03-08:
Why would I mind :-)? It's just that I fear synchronization issues (e.g., sending 6LoWPAN fragments subsequently) and thus, I'd consider this a high impact feature which needs to be tested properly. Going through the release tests would be a good way to go. The cutoff date for high impact features is 26th March.

PeterKietzmann, 2019-05-28:
This relates to ongoing work by @jia200x  which is partly described in #11473 and #11483.

PeterKietzmann, 2020-04-09:
@roberthartung #13798 got merged! an we close this issue?

closed 2020-04-09 by roberthartung