RIOT-OS/RIOT #8221: at86rf2xx: Where is CRC checking implemented?

roberthartung, 2017-12-07:
Q: FCS is ignored in the driver. So is there any check? Or does the driver expect the radio to drop packages with wrong CRC sum?
Are there any plans to add crc_ok to the radio_rx_info struct?

miri64, 2017-12-07:
> Q: FCS is ignored in the driver. So is there any check? Or does the driver expect the radio to drop packages with wrong CRC sum?

Isn't the FCS already checked by the device? I seem to remember that frames with wrong FCS were usually dropped by the device.

> Are there any plans to add crc_ok to the radio_rx_info struct?

AFAIK there are no plans for that, so if you require one, go ahead and add it.

roberthartung, 2017-12-07:
> Isn't the FCS already checked by the device? I seem to remember that frames with wrong FCS were usually dropped by the device.

In Basic mode, CRC check is not handled automatically. This is important in energy-sensitive applications, where data might still be useful.

> AFAIK there are no plans for that, so if you require one, go ahead and add it.

will do :)

miri64, 2017-12-07:
> In Basic mode, CRC check is not handled automatically. This is important in energy-sensitive applications, where data might still be useful.

I seem to remember, that @haukepetersen once told me, that only extended mode is implemented for at86rf2xx...

roberthartung, 2017-12-07:
> I seem to remember, that @haukepetersen once told me, that only extended mode is implemented for at86rf2xx...
working on getting the basic mode running. Right now my applications stops sending packets (or receiving some?). Not sure why... Gotta debug this .. somehow.

closed 2019-09-10 by stale[bot]