RIOT-OS/RIOT #8238: kinetis: linker fail to link on Raspbian

smlng, 2017-12-11:
When trying to build RIOT on Raspian Jessie with arm-none-eabi toolchain from APT (gcc 4.8.x, ld 2.25) for `BOARD=pba-d-01-kw2x` aka PhyNode there is a linker error:

```
kinetis.ld:30: nonconstant expression for origin
collect2: error: ld returned 1 exit status
```

changes where introduced in #7882 and work with more recent toolchains (e.g. >5.x.y).

smlng, 2017-12-11:
actually its not Raspbian but Debian Jessie specific, or even more specific the arm-toolchain version.

gebart, 2017-12-11:
Sorry, but I don't want to hold the development back to work around bugs or missing features in ancient toolchains. AFAIK the policy for the project is to support the latest LTS release of Ubuntu, which is 16.04 at the moment, which provides a more recent binutils than this. Does it work if you upgrade binutils to 2.26?

miri64, 2017-12-12:
The most current Debian stable is Stretch, which for the most part is more up-to-date than the current LTS of Ubuntu (don't know if raspbian was updated to Stretch yet, though).

smlng, 2017-12-12:
> don't want to hold the development back to work around bugs or missing features in ancient toolchains

> most current Debian stable is Stretch, which for the most part is more up-to-date

@gebart and @miri64: you're both right, I updated from Jessie to Stretch (Raspbian and Debian) and with the newer toolchains everything is working as expected. I also agree, that we don't need or want to support rather old toolchains. However, I think other might/will hit that issue, too. Hence, I thought it might be best to open an issue and provide an answer + solution, so other can find it here.

TL;DR: GCC-4.8.x toolchain is too old for RIOT, use GCC >= 4.9.x for either RIOT native or any board.

kYc0o, 2018-07-19:
Thus, @smlng close?