RIOT-OS/RIOT #8240: periph/rtc breaks nucleo-l152

kYc0o, 2017-12-11:
While re-testing #7712 I realised that it didn't work at all, then I tested other examples and nothing seems to work.

After bisecting and looking everywhere, I realised that #6900 was the culprit. Unfortunately I didn't have time to dig more into the issue, so I don't know exactly with commit is the real cause of the crash.

Even if I revert the changes, my PR still not work, but I'm investigating if there are more bugs related to this platform.

aabadie, 2017-12-11:
Yes nucleo-l152 doesn't work actually, as reported in #7687, see [comment](https://github.com/RIOT-OS/RIOT/pull/7687#issuecomment-343956480). Maybe your problem is related to this ?

aabadie, 2017-12-11:
Also realized that there's already an issue here: #8024. Duplicate ?

kYc0o, 2017-12-11:
Definitely this is a duplicate... however I realised that reverting #6900 actually solves the problem for the basic examples. I'll try to move the discussion in those PR/issues. Please don't close this one until that's done.

kaspar030, 2017-12-15:
duplicate

closed 2017-12-15 by kaspar030

kYc0o, 2017-12-25:
I just made the reference on #8024, I didn't want to close this without having it related to it. Now it's ok.

PR #8545 (2018-03-01): boards/nucleo-l152: configure LSI by default

### Contribution description
Some nucleo-l152 boards, namely revision C-01, don't have a LSE to rely on for low speed clocking, which is mostly used for RTC.

This PR makes it optional, and define LSI by default.

### Issues/PRs references
Fixes #8024 and fixes #8240, depends on ~#8518~ to fix everything.