RIOT-OS/RIOT #8242: at86rf2xx: Dead lock when sending while receiving

roberthartung, 2017-12-12:
The at86rf2xx driver doesn't check the current state when sending. Therefor it is possible, that a deadlock occurs. `at86rf2xx_tx_prepare` tries to set the state to `AT86RF2XX_STATE_TX_ARET_ON`, however, if the receiver is currently in `BUSY_RX` state, the idle_state will be set accordingly.
The send afterwards is still successful, however, the idle_state is wrong. The `_isr` method of `at86rf2xx_netdev.c` will set the `idle_state` afterwards, however it will never be reached and this results in a dead lock.

I will provide a fix shortly: Add busy waiting before sending / switching states.

roberthartung, 2017-12-12:
Update: Probably extending at86rf2xx_set_state to also check for BUSY_RX and BUSY_TX might solve the problem. Will update issue accordingly.

gebart, 2017-12-19:
Are you working on a PR for this or is this a potential "help wanted" issue?

kaspar030, 2018-04-16:
Any news here?

virajsahai, 2018-05-17:
If this is a "help wanted" candidate..I would like to take this up!

gebart, 2018-05-17:
@virajsahai any help is welcome. Feel free to open a PR if you find what the actual problem is.

DipSwitch, 2018-08-12:
This is the same bug as described under: https://github.com/RIOT-OS/RIOT/pull/7276#issuecomment-412254790

It has been tried to be fixed with PR's: #7115 #7275 #7276 #8186

miri64, 2020-07-01:
@roberthartung Can you check if https://github.com/RIOT-OS/RIOT/pull/12728 fixed this? I think this is the same behavior as described in https://github.com/RIOT-OS/RIOT/pull/11256.