RIOT-OS/RIOT #8265: Tracker: GCC 7.x compatibility

gebart, 2017-12-14:
#### Description
This is a tracker bug for any issues which occur only with GCC version 7.x or higher. Currently the CI system is using GCC 6.x so these issues can only be seen on custom toolchain installations.

#### Steps to reproduce the issue

 1. Install GCC 7.x, set as default compiler for your platform
 2. Run `make buildtest` or `dist/tools/compile_test/compile_test.py`
 3. Report any failed builds including a log of the error. Alternatively, provide a PR with the errors corrected.
 4. Reference this Issue number (#8265) in the description of your Issue/PR
 5. A notice will be added in this Issue automatically by Github

#### Issue/PR references

 - [ ] #8243 unittests: Fix printf float test BUFSIZE

smlng, 2018-01-16:
there lots of more issue regarding GCC 7.x, see this [Murdock-Log](https://ci.riot-os.org/RIOT-OS/RIOT/8362/4ebf00c0ff1eeab208869fba457e4e9febc15a15/output.html) where I tried to add a CI-Worker with `arm-none-eabi-gcc-7.2`. I'll work on fixes for these, help welcome. Tinydtls fix is provided in #8367

cladmi, 2018-03-15:
All currently reported issues have been closed ! Cool.

haukepetersen, 2018-05-03:
As all issues have been closed, I take the freedom to close this issue as well. Good job guys!

closed 2018-05-03 by haukepetersen