RIOT-OS/RIOT #8266: emlib: warnings when using GCC 7

gebart, 2017-12-14:
#### Description
This error occurs when building the default example for the ikea-tradfri board when using GCC 7.2.0:

#### Steps to reproduce the issue

Use GCC 7.2.0
`make BOARD=ikea-tradfri -C examples/default`

#### Expected results

No compilation errors

#### Actual results

```
/home/jgn/work/src/riot/examples/default/bin/pkg/ikea-tradfri/emlib/emlib/src/em_cmu.c: In function ‘CMU_ClockSelectSet’:
/home/jgn/work/src/riot/examples/default/bin/pkg/ikea-tradfri/emlib/emlib/src/em_cmu.c:3463:14: error: this statement may fall through [-Werror=implicit-fallthrough=]
       selReg = (selReg == NULL) ? &CMU->LFACLKSEL : selReg;
       ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/jgn/work/src/riot/examples/default/bin/pkg/ikea-tradfri/emlib/emlib/src/em_cmu.c:3480:5: note: here
     case CMU_LFECLKSEL_REG:
     ^~~~
/home/jgn/work/src/riot/examples/default/bin/pkg/ikea-tradfri/emlib/emlib/src/em_cmu.c:3481:14: error: this statement may fall through [-Werror=implicit-fallthrough=]
       selReg = (selReg == NULL) ? &CMU->LFECLKSEL : selReg;
       ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/jgn/work/src/riot/examples/default/bin/pkg/ikea-tradfri/emlib/emlib/src/em_cmu.c:3488:5: note: here
     case CMU_LFBCLKSEL_REG:
     ^~~~
cc1: all warnings being treated as errors
```

#### Versions

arm-none-eabi-gcc (Gentoo 6.4.0 p1.0) 6.4.0

miri64, 2017-12-14:
Potential fixes: either add a patch with the according comments to the switch-case (`/* falls through intentionally */`) or deactivate `-Werror=implicit-fallthrough` for `emlib`.

nmeum, 2018-01-16:
This seems to have been fixed, at least I can't reproduce this anymore with gcc 7.2.0. The switch statements contain `/* Fall through and [...] */` comments now which should stop gcc from emitting a warning.

PR #8554 (2018-02-13): pkg: gecko_sdk: update version for GCC7 fixes.
### Contribution description

This updates the Gecko SDK package (used by EFM32 CPU) to fix GCC7 fall-through errors.

### Issues/PRs references

Fixes #8266, alternative to #8487.
closed 2018-02-13 by kYc0o