RIOT-OS/RIOT #8267: docker: Default settings should provide a volume for git-cache

gebart, 2017-12-14:
#### Description
Running the build in a Docker container with the `BUILD_IN_DOCKER=1 make` command should provide a persistent volume for storing the git-cache directory to reduce network download traffic.

#### Steps to reproduce the issue

in examples/ccn-lite-relay:

```
BUILD_IN_DOCKER=1 make
```

#### Expected results
git sources should be loaded from the git-cache

#### Actual results
git sources are re-downloaded whenever the `bin` directory is deleted

#### Versions
riot/riotbuild docker image b329486bb877 from 2017-12-01T14:36:49.484Z
up-to-date master branch from 2017-12-14

cladmi, 2018-07-31:
I will look at it.
I think it would require to move the default out of `dist/tools/git/git-cache`:

```
GIT_CACHE_DIR=${GIT_CACHE_DIR:-${HOME}/.gitcache}
```

To have access to the value in the build system.

cladmi, 2018-07-31:
I proposed something in #9651 

The difference from this issue description is that you say "default settings" whereas I only did it if the ${GIT_CACHE_DIR} is defined and exists, I do not create one.