RIOT-OS/RIOT #8284: static_tests: Verify modules used in precompiler conditionals

bergzand, 2017-12-19:
#### Description
Preprocessor conditionals using modules in the expression are not checked whether the modules actually exists. We rely on the programmer/testers to verify that the module being checked actually exists. Having a static tests that verifies whether any `MODULE_SOME_MODULE` used is actually an existing module helps spotting these errors.

As an example:
`#if defined(MODULE_IPV6_ROUTER)…`

This module doesn't exist (should have been `MODULE_GNRC_IPV6_ROUTER`). 

#### Steps to reproduce the issue
I'm going to give #8101 as an example for this.

#### Expected results
Travis/Murdock spots this kind of mistakes.

#### Actual results
The application doesn't behave as expected and I have to spot these errors myself.

cladmi, 2018-01-30:
I see 3 steps for this:

* generate the list of all modules
* extract the module names used in a file
* Magic to run both and produce test result

The first one is hard to do right now at without matching too large.
Getting (real) modules names and pseudomodules as names or regexp too is not too complicated.
Having exact pseudomodules and submodules would be hard right now.
It could be enough however for a first version, right now we just accept everything so it would be an improvement.

cladmi, 2018-01-30:
Oh, and ignoring packages modules and sources as its harder to do also.

madokapeng, 2018-06-14:
Any update on this issue now. I am assigned to solve it @cladmi @bergzand

cladmi, 2018-06-14:
Nothing on my side

miri64, 2019-09-10:
I think this should be implemented.