RIOT-OS/RIOT #8285: pkg: download of packages broken?

haukepetersen, 2017-12-19:
Stumbled upon this last week when doing #8269, and now again when reviewing #5544: when building with `make -B clean all` the used package(s) are downloaded and patched twice:
```
rm -Rf /home/hauke/dev/riot/RIOT/examples/senml_json/bin/pkg/native/jsmn
mkdir -p /home/hauke/dev/riot/RIOT/examples/senml_json/bin/pkg/native/jsmn
/home/hauke/dev/riot/RIOT/dist/tools/git/git-cache clone "https://github.com/zserge/jsmn" "b77d84ba48e057aa464b6c6b6f6209e632918cb3" "/home/hauke/dev/riot/RIOT/examples/senml_json/bin/pkg/native/jsmn"
Cloning into '/home/hauke/dev/riot/RIOT/examples/senml_json/bin/pkg/native/jsmn'...
remote: Counting objects: 469, done.
remote: Total 469 (delta 0), reused 0 (delta 0), pack-reused 469
Receiving objects: 100% (469/469), 173.12 KiB | 0 bytes/s, done.
Resolving deltas: 100% (230/230), done.
Checking connectivity... done.
HEAD is now at b77d84b... Merge pull request #75 from condemned77/master
touch /home/hauke/dev/riot/RIOT/examples/senml_json/bin/pkg/native/jsmn/.git-downloaded
git -C /home/hauke/dev/riot/RIOT/examples/senml_json/bin/pkg/native/jsmn checkout -f b77d84ba48e057aa464b6c6b6f6209e632918cb3
HEAD is now at b77d84b... Merge pull request #75 from condemned77/master
git -C /home/hauke/dev/riot/RIOT/examples/senml_json/bin/pkg/native/jsmn am --ignore-whitespace "/home/hauke/dev/riot/RIOT/pkg/jsmn"/patches/*.patch
Applying: RIOTize makefile
touch /home/hauke/dev/riot/RIOT/examples/senml_json/bin/pkg/native/jsmn/.git-patched
"make" -C /home/hauke/dev/riot/RIOT/pkg/jsmn
rm -Rf /home/hauke/dev/riot/RIOT/examples/senml_json/bin/pkg/native/jsmn
mkdir -p /home/hauke/dev/riot/RIOT/examples/senml_json/bin/pkg/native/jsmn
/home/hauke/dev/riot/RIOT/dist/tools/git/git-cache clone "https://github.com/zserge/jsmn" "b77d84ba48e057aa464b6c6b6f6209e632918cb3" "/home/hauke/dev/riot/RIOT/examples/senml_json/bin/pkg/native/jsmn"
Cloning into '/home/hauke/dev/riot/RIOT/examples/senml_json/bin/pkg/native/jsmn'...
remote: Counting objects: 469, done.
remote: Total 469 (delta 0), reused 0 (delta 0), pack-reused 469
Receiving objects: 100% (469/469), 173.12 KiB | 0 bytes/s, done.
Resolving deltas: 100% (230/230), done.
Checking connectivity... done.
HEAD is now at b77d84b... Merge pull request #75 from condemned77/master
touch /home/hauke/dev/riot/RIOT/examples/senml_json/bin/pkg/native/jsmn/.git-downloaded
git -C /home/hauke/dev/riot/RIOT/examples/senml_json/bin/pkg/native/jsmn checkout -f b77d84ba48e057aa464b6c6b6f6209e632918cb3
HEAD is now at b77d84b... Merge pull request #75 from condemned77/master
git -C /home/hauke/dev/riot/RIOT/examples/senml_json/bin/pkg/native/jsmn am --ignore-whitespace "/home/hauke/dev/riot/RIOT/pkg/jsmn"/patches/*.patch
Applying: RIOTize makefile
touch /home/hauke/dev/riot/RIOT/examples/senml_json/bin/pkg/native/jsmn/.git-patched
"make" -C /home/hauke/dev/riot/RIOT/examples/senml_json/bin/pkg/native/jsmn
...
```
Seen on `native` and on `samr21-xpro`...

kaspar030, 2018-01-22:
Recursive make in combination with "-B" is the problem here:

1. First-level make remakes "prepare", which triggers "pkg-prepare", which then calls "make prepare" in each package directory, triggering a download.

2. First-level make then triggers the actual package build (using ```make -C$(PKGDIR) all```), which is, through "-B", forced to rebuild "git-download...

I *think* that's what is happening. Short solution: don't use "-B".

haukepetersen, 2018-01-24:
Works for me. I don't consider this a serious issue (aside me gotten used to use `-B` far too often), so closing this one.

closed 2018-01-24 by haukepetersen