RIOT-OS/RIOT #8289: pkg/libcoap: patching fails with newer version of git

smlng, 2017-12-19:
#### Description

While the libcoap package still works when compiled by murdock, patching fails with new versions of GIT, e.g., on Arch Linux and macOS with git 2.15.x. Using older git version, e.g., 2.11.x on Raspbian still works.

#### Steps to reproduce the issue

Try to build `tests/pkg_libcoap` with GIT version 2.15.x.

#### Expected results

builds

#### Actual results

patching fails, starting with patch 0004*

#### Versions

* GIT version 2.15.x does not work
* GIT version 2.11.x still works.

miri64, 2017-12-19:
My machine currently only has `git` 2.7 (Ubuntu 16.04)... So I can't work on this at the moment :(

smlng, 2017-12-19:
I discovered this only very recently myself, when running compile tests on macOS, but I never used libcoap with RIOT. And Murdock didn't hit this issue yet, but might in the future, when we update the worker images - so its good to be prepared, I guess.

kYc0o, 2018-01-16:
I can confirm the issue, I experienced it while testing #7737.

smlng, 2018-01-22:
fixed by #8396

closed 2018-01-22 by smlng