RIOT-OS/RIOT #8337: openmote-cc2538: wrong variable name in makefile for JLink programmer

geith, 2018-01-09:
#### Description

The _openmote-cc2538_ makefile (_boards/openmote-cc2538/Makefile.include_) uses the variable name **JLINK_FLASH_ADDR** for the flash address instead of **FLASH_ADDR** (as for example _dist/tools/jlink/jlink.sh_). As a result, the flash ROM is not written when using the JLink programmer, without reporting an error.

#### Steps to reproduce the issue

Flash any program to the _openmote-cc2538_ via JLink programmer:
```
make BOARD=openmote-cc2538 PROGRAMMER=jlink flash
```

#### Expected results

After fixing the variable name, flashing works as expected:
```diff
diff --git a/boards/openmote-cc2538/Makefile.include b/boards/openmote-cc2538/Makefile.include
index 549c7476f..ffada2313 100644
--- a/boards/openmote-cc2538/Makefile.include
+++ b/boards/openmote-cc2538/Makefile.include
@@ -12,7 +12,7 @@ export PROGRAMMER ?= cc2538-bsl
 ifeq ($(PROGRAMMER),jlink)
   # setup JLink for flashing
   export JLINK_DEVICE := cc2538sf53
-  export JLINK_FLASH_ADDR := 200000
+  export FLASH_ADDR := 200000
   export JLINK_IF := JTAG
   export TUI := 1
   include $(RIOTMAKE)/tools/jlink.inc.mk
```

```
make BOARD=openmote-cc2538 PROGRAMMER=jlink flash

### Flashing Target ###
### Flashing at address 200000 ###
...
Cortex-M3 identified.
Halting CPU for downloading file.
Downloading file [xxx]...
Comparing flash   [100%] Done.
Erasing flash     [100%] Done.
Programming flash [100%] Done.
Verifying flash   [100%] Done.
J-Link: Flash download: Flash programming performed for 1 range (79872 bytes)
J-Link: Flash download: Total time needed: 2.955s (Prepare: 0.023s, Compare: 1.204s, Erase: 0.841s, Program: 0.876s, Verify: 0.006s, Restore: 0.004s)
O.K.

Reset delay: 0 ms
Reset type NORMAL: Resets core & peripherals via SYSRESETREQ & VECTRESET bit.

Script processing completed.
```
The second line shows the correct flash address _0x200000_ and the later output indicates the flash process.

#### Actual results

```
make BOARD=openmote-cc2538 PROGRAMMER=jlink flash

### Flashing Target ###
### Flashing at address 0x0 ###
...
Cortex-M3 identified.
Halting CPU for downloading file.
Downloading file [xxx]...
O.K.

Reset delay: 0 ms
Reset type NORMAL: Resets core & peripherals via SYSRESETREQ & VECTRESET bit.

Script processing completed.
```
The second line shows the wrong flash address _0x0_ and the flash ROM is not erased/written/verified (output missing).

#### Versions

* Operating System: Ubuntu 17.10, 64 Bit
* RIOT: latest master branch

PeterKietzmann, 2018-01-17:
@geith can you provide a pull request to fix this?

geith, 2018-01-17:
PR done, but checks failed at _./dist/tools/pr_check/pr_check.sh master_:
```
	Pull request needs squashing:
	    46a64013 fixed wrong makefile variable name
```
I don't know what's wrong, the PR contains only one commit. So what can I do?

PeterKietzmann, 2018-01-17:
Let's discuss next to the code (in the PR #8375). As @aabadie already indicated the problem is you commit message. It contains "fixup" which is detected by the CI

PR #8375 (2018-01-17): fixed wrong makefile variable name

### Contribution description

Fixed makefile variable name for flash address when using JLink programmer.

### Issues/PRs references

Fixes #8337
closed 2018-01-17 by PeterKietzmann