RIOT-OS/RIOT #8339: Feature request: File system example

miri64, 2018-01-09:
Now that `vfs` supports a few file systems, it might be time to provide an example of how to access files in the `examples/` directory. I know that you can just use the newlib syscalls, so I think that is something we should showcase.

aabadie, 2018-01-10:
It would be great if this example was using the `sd_card_spi` driver. It could be a good data logger example for Adafruit Feather-M0 and Arduino MKRZERO boards.

miri64, 2018-01-10:
I don't think its a good idea to have the example use a specific device, especially since flash-based storages are way more common in IoT scenarios.

MichelRottleuthner, 2018-01-10:
I agree with @miri64 that we should not tie the example to a specific device. But it won't hurt to keep an eye on making it easy to switch the example to use *some_storage_dev_supported_by_riot*.
>...especially since flash-based storages are way more common in IoT scenarios.

the last time I checked sd-cards were flash based ;)

miri64, 2018-01-10:
> the last time I checked sd-cards were flash based ;)

I meant the thingies you solder to your board ;-)

PR #8559 (2018-04-13): examples: add filesystem example
### Contribution description

This is a basic example showing how to use file systems within RIOT.
Some shell commands are provided to demonstrate mount/format/unmount and read/write in file.
Tested on native with littlefs and spiffs.

### Issues/PRs references

Fix #8339
closed 2018-04-13 by kaspar030