RIOT-OS/RIOT #8340: telosb/MSP430 startup serial preamble

rienafairefr, 2018-01-09:
Hey folks,

I'm working on IoT-lab at INRIA on including support for new custom boards. [see here](https://www.iot-lab.info/hardware/custom/)

Among the new boards where the development is in progress is a telosb/TmoteSky, based on MSP430 MCU
We have a basic 'autotest' RIOT application, nothing fancy and it works fine for other boards we have deployed. But on the TmoteSky I've encountered an issue on the serial port communication, and I've successfully made it work, and it basically boils down to the startup code in between board_init and kernel_init [here](https://github.com/RIOT-OS/RIOT/blob/1bf222e07ff9aac0c3043dc0b440c6da8c9e5667/cpu/msp430_common/startup.c#L35)

Just after flashing the board, the gateway on which the node is plugged expects to have the serial link open and clear (one of the basic autotest is sending `echo msg` and expecting `msg` in return), and this startup code is "in the pipe" so that autotest fails.

There is also on top of that (same kind of issue), the startup LOG_INFO in kernel_init [here](https://github.com/RIOT-OS/RIOT/blob/93ce5a7e49096db1468b56e3aa500e1153c23b83/core/kernel_init.c#L57), but this is easily solvable, e.g. by compiling with -DLOG_LEVEL=LOG_NONE

#### Conclusion

The line with the `put`s is easily removed before compiling the firmware and it works. Is it desirable to have this configurable at compile, I can send a PR ?

aabadie, 2018-01-10:
> Is it desirable to have this configurable at compile, I can send a PR ?

What do you mean ? using `LOG_INFO` like it is done in `kernel_init` ? and deactivate the after `board_init` message at compile time.

If yes, it sounds ok for me.

PeterKietzmann, 2018-01-11:
In my opinion this shouldn't be a puts anyway. I would either remove it completely or make it a logging message, as proposed...

rienafairefr, 2018-01-11:
Yes, I think the simplest would be puts => LOG_INFO. see PR#8347 just now

PeterKietzmann, 2018-01-11:
Fixed with #8347 ? Can we close this issue?

rienafairefr, 2018-01-11:
Yes, fixed, closing it :-) Thanks to all

closed 2018-01-11 by rienafairefr

rienafairefr, 2018-01-11:
OK for me. Closing