RIOT-OS/RIOT #8356: llvm: Build broken with clang 5.0.0

pyropeter, 2018-01-12:
Linking breaks with current clang:

```
$ make TOOLCHAIN=llvm
Building application "$app" for "bluepill" with MCU "stm32f1".

"make" -C /boards/bluepill
"make" -C /core
"make" -C /cpu/stm32f1
"make" -C /drivers
"make" -C /sys
"make" -C /cpu/cortexm_common
"make" -C /sys/auto_init
"make" -C /drivers/periph_common
"make" -C /cpu/cortexm_common/periph
"make" -C /cpu/stm32_common
"make" -C /sys/div
"make" -C /cpu/stm32_common/periph
"make" -C /sys/isrpipe
"make" -C /sys/newlib_syscalls_default
"make" -C /sys/pm_layered
"make" -C /sys/tsrb
"make" -C /sys/uart_stdio
"make" -C /sys/xtimer
"make" -C /cpu/stm32f1/periph
clang-5.0: warning: argument unused during compilation: '-nostartfiles' [-Wunused-command-line-argument]
clang-5.0: warning: argument unused during compilation: '-specs=nano.specs' [-Wunused-command-line-argument]
/usr/bin/ld.lld: error: unable to find library -lm
/usr/bin/ld.lld: error: unable to find library -lgcc
/usr/bin/ld.lld: error: unable to find library -lc
/usr/bin/ld.lld: error: unable to find library -lnosys
/usr/bin/ld.lld: error: cannot open stm32_common.ld: No such file or directory
/usr/bin/ld.lld: error: unable to find library -lc
/usr/bin/ld.lld: error: unable to find library -lm
/usr/bin/ld.lld: error: unable to find library -lclang_rt.builtins-arm.a
clang-5.0: error: ld.lld command failed with exit code 1 (use -v to see invocation)
make: *** [/examples/splitflap/../../Makefile.include:321: link] Error 1
```

gebart, 2018-01-12:
I can confirm the same issue on my machine. I did not have time to investigate further, but it seems to be related to the cross toolchain path configuration makefile script.

nmeum, 2018-01-15:
Building RIOT applications with `BOARD=native` works fine though.

A simple workaround is using gcc instead of clang for linking in `toolchain/llvm.inc.mk`:

```diff
diff --git a/makefiles/toolchain/llvm.inc.mk b/makefiles/toolchain/llvm.inc.mk
index ffc38ed0e..72856ca4f 100644
--- a/makefiles/toolchain/llvm.inc.mk
+++ b/makefiles/toolchain/llvm.inc.mk
@@ -9,7 +9,7 @@ endif
 export CC          = clang
 export CXX         = clang++
 export CCAS       ?= $(CC)
-export LINK        = $(CC)
+export LINK        = $(PREFIX)gcc
 export AS          = $(LLVMPREFIX)as
 export AR          = $(LLVMPREFIX)ar
 export NM          = $(LLVMPREFIX)nm
@@ -52,7 +52,6 @@ ifneq (,$(TARGET_ARCH))
   # Tell clang to cross compile
   export CFLAGS     += -target $(TARGET_ARCH)
   export CXXFLAGS   += -target $(TARGET_ARCH)
-  export LINKFLAGS  += -target $(TARGET_ARCH)
 
   # Use the wildcard Makefile function to search for existing directories matching
   # the patterns above. We use the -isystem gcc/clang argument to add the include
```

PR #8365 (2018-02-21): makefiles: always link with GCC even if TOOLCHAIN is set to LLVM
## Contribution description

Cross-compiling RIOT applications with `TOOLCHAIN=llvm` currently doesn't work with clang-5.0 (see #8356). The reason for this being that clang-5.0 seems to try using the LLVM linker (LLD) instead of GCC for linking cross compiled object files. I don't know though why LLVM changed this behavior in clang-5.0.

A simple workaround for this was proposed in #8356 by myself. The workround is to set the `LINK` variable to GCC. After some considerations I honestly believe that this is the easiest and best so [... cut]
commit 18f99ef7 (2018-02-21): makefiles: always link with gcc even if TOOLCHAIN is set to LLVM

Fixes #8356
closed 2018-02-21 by gebart