RIOT-OS/RIOT #8369: Build failed : recipe for target 'link' failed

eli0tc, 2018-01-16:
#### Description
Wether I try to build the examples for the m3 node (following the tutorial [https://www.iot-lab.info/tutorials/riot-compilation/](https://www.iot-lab.info/tutorials/riot-compilation/)) on the Strasbourg site or the Grenoble site, the build fails and I get an error message.
It worked without problem one week ago.

#### Steps to reproduce the issue
As stated in the tutorial:
`<login>@<site>:~/iot-lab/parts/RIOT/examples/default$ BOARD=iotlab-m3 make all`

#### Actual results
```
chauvine@strasbourg:~/iot-lab/parts/RIOT/examples/default$ BOARD=iotlab-m3 make all
Building application "default" for "iotlab-m3" with MCU "stm32f1".
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/boards/iotlab-m3
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/boards/common/iotlab
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/core
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/cpu/stm32f1
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/cpu/cortexm_common
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/cpu/cortexm_common/periph
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/cpu/stm32_common
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/cpu/stm32_common/periph
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/cpu/stm32f1/periph
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/drivers
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/drivers/at86rf2xx
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/drivers/isl29020
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/drivers/l3g4200d
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/drivers/lps331ap
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/drivers/lsm303dlhc
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/drivers/netdev_ieee802154
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/drivers/periph_common
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/drivers/saul
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/sys
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/sys/auto_init
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/sys/auto_init/netif
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/sys/auto_init/saul
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/sys/div
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/sys/fmt
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/sys/isrpipe
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/sys/luid
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/sys/net/gnrc
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/sys/net/gnrc/netapi
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/sys/net/gnrc/netif
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/sys/net/gnrc/netif/hdr
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/sys/net/gnrc/netreg
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/sys/net/gnrc/pkt
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/sys/net/gnrc/pktbuf
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/sys/net/gnrc/pktbuf_static
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/sys/net/gnrc/pktdump
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/sys/net/link_layer/ieee802154
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/sys/newlib_syscalls_default
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/sys/od
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/sys/phydat
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/sys/pm_layered
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/sys/ps
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/sys/saul_reg
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/sys/shell
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/sys/shell/commands
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/sys/tsrb
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/sys/uart_stdio
"make" -C /senslab/users/chauvine/iot-lab/parts/RIOT/sys/xtimer
stm32_common.ld:25: nonconstant expression for length
collect2: error: ld returned 1 exit status
/senslab/users/chauvine/iot-lab/parts/RIOT/examples/default/../../Makefile.include:320: recipe for target 'link' failed
make: *** [link] Error 1
```

#### Versions
Build environment: Grenoble and Strasbourg site ssh

gebart, 2018-01-16:
This is a bug/missing feature in older binutils versions. Is it possible to upgrade the arm toolchain there?
Could you post the version of the tools installed? 
Use the script in RIOT/dist/tools/ci/print_toolchain_versions.sh

eli0tc, 2018-01-16:
I tried an apt-get install build-essential with no luck ; I don't have the right to do so.
Here is the output of print_toolchain_versions.sh:

```
Installed toolchain versions
----------------------------
          native gcc: gcc (Debian 4.9.2-10) 4.9.2
          msp430-gcc: msp430-gcc (GCC) 3.2.3
             avr-gcc: missing/error
   arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.9.3 20150303 (release) [ARM/embedded-4_9-branch revision 221220]
    mips-mti-elf-gcc: missing/error
./print_toolchain_versions.sh: 92: ./print_toolchain_versions.sh: clang: not found
arm-none-eabi-newlib: "2.2.0"
 mips-mti-elf-newlib: missing/error
            avr-libc: missing/error (missing/error)
            cppcheck: missing
          coccinelle: missing
                 git: git version 2.1.4
```

aabadie, 2018-01-17:
The GCC ARM toolchains on the iot-lab frontends (this is a global issue there) are a bit outdated. I'm looking into this to see how we can update them.

In the mean time, you can try building against the latest stable version: 2017.10, instead of master, in step 2. use:
```
     git clone -b 2017.10-branch https://github.com/RIOT-OS/RIOT.git parts/RIOT
```

aabadie, 2018-01-17:
Ok, I pushed the fix for the `make setup-riot` used in step 2.

Regarding the update of the toolchain on iot-lab, this is under discussion in the maintenance team. Because IoT-LAB provides support for several OSes, including RIOT, Contiki and TinyOS, we have to think to a good solution and do some testing before.

Also, this problem is only IoT-Lab related, so next time please send an email to admin@iot-lab.info or open an issue on the IoT-LAB github project: https://github.com/iot-lab/iot-lab/issues.

closed 2018-01-17 by aabadie