RIOT-OS/RIOT #8372: Native broken on OS X

kYc0o, 2018-01-16:
#### Description

Any application built in OS X fails due to some warnings.
#### Steps to reproduce the issue

Compile any example in OS X by doing `BOARD=native make`
#### Expected results

It compiles
#### Actual results

```console
/Users/facosta/git/RIOT2/RIOT/core/sched.c:95:11: error: format specifies type 'short' but the argument has type 'int' [-Werror,-Wformat]
          (active_thread == NULL) ? KERNEL_PID_UNDEF : active_thread->pid,
~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/facosta/git/RIOT2/RIOT/core/include/debug.h:97:50: note: expanded from macro 'DEBUG'
#define DEBUG(...) if (ENABLE_DEBUG) DEBUG_PRINT(__VA_ARGS__)
                                     ~~~~~~~~~~~~^~~~~~~~~~~~
/Users/facosta/git/RIOT2/RIOT/core/include/debug.h:56:20: note: expanded from macro 'DEBUG_PRINT'
            printf(__VA_ARGS__); \
                   ^~~~~~~~~~~
/Users/facosta/git/RIOT2/RIOT/core/sched.c:166:33: error: format specifies type 'unsigned short' but the argument has type 'uint8_t'
      (aka 'unsigned char') [-Werror,-Wformat]
                  process->pid, process->priority);
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~
/Users/facosta/git/RIOT2/RIOT/core/include/debug.h:97:50: note: expanded from macro 'DEBUG'
#define DEBUG(...) if (ENABLE_DEBUG) DEBUG_PRINT(__VA_ARGS__)
                                     ~~~~~~~~~~~~^~~~~~~~~~~
/Users/facosta/git/RIOT2/RIOT/core/include/debug.h:56:20: note: expanded from macro 'DEBUG_PRINT'
            printf(__VA_ARGS__); \
                   ^~~~~~~~~~~
/Users/facosta/git/RIOT2/RIOT/core/sched.c:174:33: error: format specifies type 'unsigned short' but the argument has type 'uint8_t'
      (aka 'unsigned char') [-Werror,-Wformat]
                  process->pid, process->priority);
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~
/Users/facosta/git/RIOT2/RIOT/core/include/debug.h:97:50: note: expanded from macro 'DEBUG'
#define DEBUG(...) if (ENABLE_DEBUG) DEBUG_PRINT(__VA_ARGS__)
                                     ~~~~~~~~~~~~^~~~~~~~~~~
/Users/facosta/git/RIOT2/RIOT/core/include/debug.h:56:20: note: expanded from macro 'DEBUG_PRINT'
            printf(__VA_ARGS__); \
                   ^~~~~~~~~~~
3 errors generated.
```
#### Versions

```console
Apple LLVM version 8.0.0 (clang-800.0.42.1)
Target: x86_64-apple-darwin15.6.0
Thread model: posix
```

nmeum, 2018-01-16:
This is probably fixed by #8364.

kYc0o, 2018-01-16:
@nmeum thanks! I'll test it asap.

smlng, 2018-01-23:
@kYc0o should be fixed, can you (re)test and close this one.

smlng, 2018-01-29:
I just ran `BOARDS=native ./dist/tools/compile_tests/compile_tests.py` using _2018.01-RC2_ on macOS (10.13.3) with some required tools (e.g. cmake) installed via Homebrew, and it completed with full success. So I guess we can close this (for now)?

kYc0o, 2018-01-29:
Yes we can!

closed 2018-01-29 by kYc0o