RIOT-OS/RIOT #8408: pkg/fatfs: linker error when build tests/pkg_fatfs_vfs for msb430 based boards

smlng, 2018-01-23:
#### Description

linker error when build tests/pkg_fatfs_vfs for msb430 based boards

#### Steps to reproduce the issue

Build `tests/pkg_fatfs_vfs` for any board that uses `msb430-gcc`, e.g `z1` or `telosb`.

#### Expected results

Builds.

#### Actual results

Linker error:

```
root@9410057a06eb:~/tests/pkg_fatfs_vfs# rm -rf bin; BOARD=z1 make                            
Building application "pkg_fatfs_vfs" for "z1" with MCU "msp430fxyz".

rm -Rf /data/riotbuild/tests/pkg_fatfs_vfs/bin/pkg/z1/fatfs
mkdir -p /data/riotbuild/tests/pkg_fatfs_vfs/bin/pkg/z1/fatfs
/data/riotbuild/dist/tools/git/git-cache clone "https://github.com/MichelRottleuthner/FatFs_for_RIOT.git" "61fd6ae3815170bf7bf6121f33f1ef68c2b11599" "/data/riotbuild/tests/pkg_fatfs_vfs/bin/pkg/z1/fatfs"
Cloning into '/data/riotbuild/tests/pkg_fatfs_vfs/bin/pkg/z1/fatfs'...
remote: Counting objects: 172, done.
remote: Total 172 (delta 0), reused 0 (delta 0), pack-reused 172
Receiving objects: 100% (172/172), 2.30 MiB | 3.01 MiB/s, done.
Resolving deltas: 100% (60/60), done.
Checking connectivity... done.
HEAD is now at 61fd6ae... remove dummy implementation diskio.c
touch /data/riotbuild/tests/pkg_fatfs_vfs/bin/pkg/z1/fatfs/.git-downloaded
"make" -C /data/riotbuild/pkg/fatfs
"make" -C /data/riotbuild/tests/pkg_fatfs_vfs/bin/pkg/z1/fatfs
"make" -C /data/riotbuild/boards/z1
"make" -C /data/riotbuild/core
"make" -C /data/riotbuild/cpu/msp430fxyz
"make" -C /data/riotbuild/cpu/msp430_common
"make" -C /data/riotbuild/cpu/msp430_common/periph
"make" -C /data/riotbuild/cpu/msp430fxyz/periph
"make" -C /data/riotbuild/drivers
"make" -C /data/riotbuild/drivers/mtd
"make" -C /data/riotbuild/drivers/mtd_sdcard
"make" -C /data/riotbuild/drivers/periph_common
"make" -C /data/riotbuild/drivers/sdcard_spi
"make" -C /data/riotbuild/pkg/fatfs/fatfs_diskio/mtd
"make" -C /data/riotbuild/pkg/fatfs/fatfs_vfs
"make" -C /data/riotbuild/sys
"make" -C /data/riotbuild/sys/auto_init
"make" -C /data/riotbuild/sys/auto_init/storage
"make" -C /data/riotbuild/sys/div
"make" -C /data/riotbuild/sys/isrpipe
"make" -C /data/riotbuild/sys/oneway-malloc
"make" -C /data/riotbuild/sys/tsrb
"make" -C /data/riotbuild/sys/uart_stdio
"make" -C /data/riotbuild/sys/vfs
"make" -C /data/riotbuild/sys/xtimer
"make" -C /data/riotbuild/tests/pkg_fatfs_vfs/bin/pkg/z1/fatfs
/data/riotbuild/tests/pkg_fatfs_vfs/bin/z1/fatfs.a(ff.o):(.debug_loc+0x11c3): relocation truncated to fit: R_MSP430_16_BYTE against `no symbol'
collect2: ld returned 1 exit status
/data/riotbuild/tests/pkg_fatfs_vfs/../../Makefile.include:320: recipe for target 'link' failed
make: *** [link] Error 1
```

#### Versions

RIOT docker image, with Ubuntu Xenial and msb430-gcc from `apt`:

```
msp430-gcc (GCC) 4.6.3 20120301 (mspgcc LTS 20120406 unpatched)
```

MichelRottleuthner, 2018-01-23:
Hmm the reference to `.debug_loc+0x11c3` looks a bit wierd to me. Shouldn't this get stripped away by default (i.e. when compiling without debug helpers)?
When looking at ff.o it looks like a lot of memory is wasted for the various debug sections:
```
[tests/pkg_fatfs_vfs]$ msp430-size -A -d bin/z1/fatfs/ff.o
bin/z1/fatfs/ff.o  :
section                        size   addr
.text                             0      0
<...a lot of other sections...>
.debug_frame                   1298      0
.debug_info                   10046      0
.debug_abbrev                   677      0
.debug_loc                    12807      0
.debug_aranges                  188      0
.debug_ranges                   280      0
.debug_line                    3548      0
.debug_str                     1657      0
Total                         43600
```
Building with `LINKFLAGS=-s BOARD=z1 make` works for me (also on msp430-gcc 4.6.3 20120301).

gebart, 2018-01-23:
@MichelRottleuthner The .debug_xxx sections will exist in the .o files and the .elf file because we are building with debug symbols by default (`gcc -g`), but they will not take up any space in the raw binary because they are not linked into ROM/RAM. You can check that the .bin/.hex files should have the same size regardless if you strip the .o and .elf files in advance (using the `strip` tool for example) or not.

If you use the size tool without flags on the .elf file you will get a .text, .data, .bss size summary which can be directly related to the ROM/RAM requirements: the ROM requirement is the size of the .text and .data sections combined, RAM is the size of the .data and .bss sections combined.

gebart, 2018-01-23:
Regarding this particular error, I assume this is related to issues in the aging mspgcc toolchain from 2012.

MichelRottleuthner, 2018-01-23:
@gebart thanks for clarifying this!
>...but they will not take up any space in the raw binary because they are not linked into ROM/RAM

That's basically what I assumed and what lead me to the point, that It's weird the debug sections cause the linking to fail.

Do you know if this issue is also present in the msp430 toolchain supported by TI?

cladmi, 2018-07-31:
This should now be fixed now that #9507 is merged.

The fix I merged was disabling the tests for now.

cladmi, 2018-08-24:
Some infos that it could be a toolchain issue: http://msp430-gcc-users.1086195.n5.nabble.com/Linker-error-with-binutils-2-22-msp20120406-2-td6527.html fixes in gcc 4.7

kaspar030, 2019-10-19:
Fixed by #12457.

closed 2020-08-05 by maribu