RIOT-OS/RIOT #8415: Where is gnrc_netif_get ?

astralien3000, 2018-01-23:
Hello,

[An old pull request is using `gnrc_netif_get`](https://github.com/named-data-iot/ndn-riot/blob/7e4f3e0a1b914239493adf78c11e96354f595e24/netif.c#L43).
I guess the API has changed.

How can I replace this piece of code ?

miri64, 2018-01-23:
Rather than just being represented by there PIDs and leaving everything else to the upper layers, network interfaces are now represented by structs since #7925. You can use [`gnrc_netif_iter()`](http://doc.riot-os.org/group__net__gnrc__netif.html#gaa58a468fb9e82d7107e229f0239c4e53) to get the interfaces and [`gnrc_netif_numof()`](http://doc.riot-os.org/group__net__gnrc__netif.html#ga79b293e98896c2831c2ca6750b32c031) to get the number of interfaces.

miri64, 2018-01-23:
(the PIDs which were previously stored in `ifs` you can access via the `gnrc_netif_t::pid` member)

miri64, 2018-01-29:
@astralien3000 I hope I was able to answer your question. I'm closing this issue now.

closed 2018-01-29 by miri64