RIOT-OS/RIOT #8417: Problem ping6

mm-dimmiz, 2018-01-23:
#### Description
Hello Guys,

We have ported riot-os to the XMC4500. And now we are trying to port the eth-driver to the xmc4500. We think that we may have actually achieved it. But now we have the a problem with the test. 

Our test application is basiclly the same as the tests/driver_encx24j600. Where we can ping over the serial console. And here comes the problem. In the console.log file you can see the send and received packages from the ethernet driver and in the wireshark.log you can see the send packages. Everything seems fine to us but still we get some strange output from the ping6. Any guesses what the problem could be?

#### Actual results
see attached files:
[console.log](https://github.com/RIOT-OS/RIOT/files/1656518/console.log)
[wireshark.log](https://github.com/RIOT-OS/RIOT/files/1656519/wireshark.log)

#### Versions
Operating system: Windows 10
Build environment: MinGW with arm-noneabi-gcc

miri64, 2018-01-23:
Hi, can I have a look at your implementation? Maybe that could already help with finding the issue. Blind I can only guess. Is your receive buffer cleared properly? The output hints at packet duplication, so maybe the receive function of netdev is returning the same packet over and over again?

miri64, 2018-01-23:
(also maybe use `txtsnd` instead so you can debug first on the link layer before putting IPv6 on top [see `tests/driver_xbee` as an example how to do that).

mm-dimmiz, 2018-01-24:
Sorry I can't provide you with the implementation :(. But the receive buffers are cleared properly as you can see in the console.log. Every frame which is printed to the console is moved to the higher layer. Also the dupplicated frame makes no sense as you can see here:

send:
00 E0 4C 13 59 0C 01 02 45 19 00 00 86 DD 60 00 
00 00 00 0C 3A 40 FE 80 00 00 00 00 00 00 CA FE 
CA FE CA FE 00 01 FE 80 00 00 00 00 00 00 CA FE 
CA FE CA FE 00 02 [ 80 00 19 C1 00 53 00 01 53 53 
53 53 ]

received: 
01 02 45 19 00 00 00 E0 4C 13 59 0C 86 DD 60 00 
00 00 00 0C 3A 80 FE 80 00 00 00 00 00 00 CA FE 
CA FE CA FE 00 02 FE 80 00 00 00 00 00 00 CA FE 
CA FE CA FE 00 01 [ 81 00 18 C1 00 53 00 01 53 53 
53 53 ]

ping timeout
dropping additional response packet (probably caused by duplicates)
dupplicated: 
81 00 18 C1 00 53 00 01 53 53 53 53

I only receive one frame with this payload and the programms says that it is dupplicated. In the _recv function i only take the buf pointer which is provided by the higher stack and copy the received frame to it. 

The next thing that i do not understand is that i get a timeout. Even if I do set the ping timeout to 10 seconds I do get a ping timeout. 

As you have suggested I have testet the drivers_xbee example and it looks just fine. I can send strings with txtsnd.

miri64, 2018-01-24:
> As you have suggested I have testet the drivers_xbee example and it looks just fine. I can send strings with txtsnd.

Did you also get the correct receive output at the other end? I'm looking into the rest tomorrow.

miri64, 2018-01-25:
> Also the dupplicated frame makes no sense as you can see here

I honestly don't see it how the dump would make sense of that. On which end and on which layer is send dumped? On which layer is receive dumped (I suppose [somewhere here in gnrc_netif_ethernet.c](https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/netif/gnrc_netif_ethernet.c#L185-L197), but I'm not sure)?

> I only receive one frame with this payload and the programms says that it is dupplicated. In the _recv function i only take the buf pointer which is provided by the higher stack and copy the received frame to it.

But you also follow the rules, for when `buf == NULL` and `len == 0` (return number of bytes currently in buffer) and `buf == NULL` and `len > 0` (flush buffer)? See [doc](http://doc.riot-os.org/structnetdev__driver.html#ae2c8cad80067e3b1f9979931ddb3cc8b) for that.

> The next thing that i do not understand is that i get a timeout. Even if I do set the ping timeout to 10 seconds I do get a ping timeout.

I suppose, but am not sure that is because the datagrams come in out of order.

mm-dimmiz, 2018-01-26:
> I honestly don't see it how the dump would make sense of that. On which end and on which layer is send dumped? On which layer is receive dumped (I suppose somewhere here in gnrc_netif_ethernet.c, but I'm not sure)?

As suggested by riot doc i have implemtend the netdev_driver_t struct which then calls the _recv function which then calls the internal receive function of the driver. This is where the hex dump comes from. 

> But you also follow the rules, for when buf == NULL and len == 0 (return number of bytes currently in buffer) and buf == NULL and len > 0 (flush buffer)? See doc for that.

If I understand it correctlly my low level receive function should do the obove mentiond procedure? If so than this is not the case at the moment.

miri64, 2018-01-26:
> If I understand it correctlly my low level receive function should do the obove mentiond procedure?

Depends on what you mean by low-level, but yes, the `netdev_driver_t::recv` method should at least do the above mentioned procedure.

closed 2020-03-13 by stale[bot]