RIOT-OS/RIOT #8424: openocd.sh: line 195: setsid: command not found

kYc0o, 2018-01-23:
#### Description

While running openocd for debugging (e.g. make debug) on OS X I have the following:

```console
openocd.sh: line 195: setsid: command not found
```

Not sure if it's an issue but I wonder if the command `setsid` is strictly necessary.

#### Steps to reproduce the issue

Run `make debug` on a fresh installed OS X.
#### Expected results

No errors.
#### Actual results

Error described above.
#### Versions

```console
Installed toolchain versions
----------------------------
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
          native gcc: Apple LLVM version 8.0.0 (clang-800.0.42.1)
          msp430-gcc: msp430-gcc (GCC) 4.7.0 20120322 (mspgcc dev 20120911)
             avr-gcc: avr-gcc (GCC) 7.2.0
   arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
    mips-mti-elf-gcc: missing/error
               clang: Apple LLVM version 8.0.0 (clang-800.0.42.1)
arm-none-eabi-newlib: "2.5.0"
 mips-mti-elf-newlib: missing/error
            avr-libc: "2.0.0" ("20150208")
            cppcheck: missing
          coccinelle: missing
                 git: git version 2.16.0
```

BTW the command cited to give the versions doesn't print openocd...

gebart, 2018-01-24:
There's a comment in the openocd script about setsid: "The setsid command is needed so that Ctrl+C in GDB doesn't kill Openocd"

gebart, 2018-01-24:
Good point about the versions script and openocd

kYc0o, 2018-01-24:
> There's a comment in the openocd script about setsid: "The setsid command is needed so that Ctrl+C in GDB doesn't kill Openocd"

So there's no way to avoid it? Something more standard? I was reading that this function might seem POSIX or mandatory for POSIX systems but apparently not, so it's not budled in OS X. The workaround I used in the past was to compile a kind of similar version [here](https://github.com/jerrykuch/ersatz-setsid/blob/master/setsid.c) but I thought that maybe we could find an alternative.

aabadie, 2018-10-19:
@kYc0o, maybe #10204 could help ?

aabadie, 2019-08-11:
Fixed by #10204

PR #10204 (2019-10-03): tools: build and use setsid command from external source when not available

### Contribution description

This PRs tries to fix #8424 by introducing some make logic to fetch, build and use setsid from an external repository. Apparently setsid is not available by default on MacOSX.

The Make logic in dist/tools and makefiles should be ok. I'm not sure whether the changes in Makefile.include could be improved.

Maybe the external repository is not the best one but this can be changed easily.

### Testing procedure

1. Start a debug session on Linux (which provides setsid) using the `debug` targ [... cut]
closed 2019-10-03 by aabadie