RIOT-OS/RIOT #8434: Lua support

danpetry, 2018-01-25:
#### Description

There is an external developer who wants to build a modular piece of hardware with a coprocessor running RIOT, with potential to be a commercial product. For this, he wishes to use Lua to control part of it, and therefore Lua support could enable commercial use of RIOT. Additionally, dynamic language support (see also https://github.com/RIOT-OS/RIOT/pull/7796 and https://github.com/RIOT-OS/RIOT/pull/2968) could be an approach to achieve over-the-air update of an application and a multi-application development paradigm for the RAPstore project (RIOT app store).

There is a benefit/cost tradeoff: the above description is the benefit, and time spent to implement Lua support (and eg not develop other features) is the cost. Therefore part of the earlier stages of development will be to judge the total amount of time that would be required.

Currently the most promising vector of approach seems to be to use the [LuaRTOS](https://github.com/whitecatboard/Lua-RTOS-ESP32) project, where the Lua engine has been implemented on top of FreeRTOS, including a range of API bindings we would need to implement such as I2C, ADCs, and LoRAWAN.

waehlisch, 2018-01-25:
@danpetry it is probably helpful to enumerate the other Lua implementions that you looked on.

jia200x, 2018-01-30:
> Lua is an interpreted language, which inherently requires dynamic allocation, in case e.g. callbacks are redefined during runtime. We try and avoid dynamic memory allocation as far as possible. We should keep an eye on how Lua handles this without potentally running out of heap space.

There are some libraries (e.g Jerryscript) that provide [custom dynamic allocation procedures](https://github.com/jerryscript-project/jerryscript/tree/master/jerry-core/jmem). Maybe LuaRTOS is doing the same.

Otherwise, I think it would be possible to use TLSF

danpetry, 2018-02-16:
Here are my findings:

# Architecture/implementation

LuaRTOS, for the most part, simply implements the original Lua library. It adds this, FreeRTOS, and the SDK for the lower layers of the ESP32 board together, along with an assortment of other libraries that the originating company (Whitecat) has deemed useful.

They have also added support for Lua multithreading, where multiple Lua processes can share the same Lua state and communicate and synchronize etc. This doesn't seem like something we will need in the first implementation, depending on the requirements of the guy who's driving this (see above).

The Lua engine itself provides an interpreter which it looks like is designed to be easily attached to a thread. It also provides bindings for whatever memory allocation mechanism you wish to use. I am going to proceed with TLSF.

# Size

Currently the Lua library (v5.3.4) when compiled using the RIOT build system takes up 86 kB of ROM and 64B of static RAM. The tech note referenced above said that a 64 kB build (v4.0) can be reduced to 24 kB of ROM by omitting the following:

- Any standard libraries that are not being used for the particular application
- The lexer, parser, and code generator; thereby making it possible to only interpret pre-compiled Lua code. (most of the gains are from this).

For v5.3.4 compiled using the RIOT build system this corresponds to a potential ROM size of ~32 kB.

danpetry, 2018-03-28:
Some further notes:

### Getting it working with a simple example

**http://lua-users.org/wiki/BindingCodeToLua**

This gives some info on how to do it and links to the lua API documentation.

**Kaspar's jerryscript PR: https://github.com/RIOT-OS/RIOT/pull/7796**

This is using the event queue in PR 6000

**https://github.com/RIOT-OS/RIOT/pull/7794**

this is a simple print function that prints out from a js file. Basically has a main.js in the example. Then does some stuff in the C program. Good idea to use this structure as a basis for the first binding (getting a simple print to work). Converts it into a char array in bin/native/js

**The lua cli program**

If handed a lua file, it runs this. If no arguments, it runs a cli. For the first, argc is 2, argv[0] is the name of the program as typed in (e.g. "./lua") and argv[1] is a string with the name of the file as typed in (e.g. ./helloworld.lua or helloworld.lua, or however it was typed in).

### Creating bindings

**http://www.lua.org/notes/ltn003.html**

The above gives an interesting approach to implementing system calls in Lua. There's a summary at the bottom, and I didn't read it thoroughly, but it sounds like a generic way to interface between Lua code and system calls, with any data and function you choose translated between Lua and the system call, and then a simple wrapper around this which is different for each specific system call. Sounds like it could be an easy and clean way to implement all the bindings. Could also be unsafe though.

**http://webserver2.tecgraf.puc-rio.br/~celes/tolua/**

The above is called tolua and is a way of automatically generating bindings. Sounds like you still need to tell it where the info for the bindings are, i.e. need to configure it properly. This seems like the most obvious way, whereas the above is a more clever way. (The above also feels cleaner and more extensible.)

 
**API doc for lua_CFunction**

tells you how to write functions that communicate with Lua.

danpetry, 2018-03-28:
As of submitting https://github.com/RIOT-OS/RIOT/pull/8847 there are two known bugs:

- [ ] user can't exit from the lua prompt back to the RIOT shell on boards
- [ ] the lua prompt on boards only appears for the results of commands

danpetry, 2018-05-09:
# Tests - WIP

The testing and support is split into "initial" and "later" using the rationale that the items in "initial" should satisfy the requirements of the product being built running RIOT, and "later" should expand this support. Also, the items in "initial" will allow us to gain familiarity with Lua idoms and good practices in low-resource dynamic language support in general.

## Initial

- [Lua test suite](https://www.lua.org/tests/), cut down to remove functions we're not supporting, should run on the "Initial" supported boards

- Integration tests for each of the APIs below should be done by having a Lua script which runs through the API calls, and possibly using pyexpect statements on a host computer connected to a RIOT node, we assert that the end effect happens - sensor/actuator behaviour, data transmitted, etc. The tests should be run on desk and written in order to integrate with the future HIL setup as far as possible.

- SAUL API functions and their calls from Lua should look like and behave like the following:
  - [TBD]
- Sock API functions and their calls from Lua should look like and behave like the following:
  - [TBD]
- Serial API functions and their calls from Lua should look like and behave like the following:
  - [TBD]
- Application loading and running from remote flash should function as the following:
  - [TBD]

## Later

- More RIOT APIs to be supported (TBD)
- Same Lua test suite as above should run on the "Later" supported boards
- Unit testing by generating stubs for each API or callbacks which return known values, with a Lua script which runs through the API calls, and Embunit to run the tests and integrate with the current Murdock setup.
- Stress tests - run for a few days on a loop
- Stress tests - run a number of times
- Record memory usage for each test (all kinds)

# Supported platforms

## Initial

- Native
- PIC32MK1024GPD100
- A high resource, popular ARM board currently supported by RIOT

## Later

- Expand support to popular, lower resource boards (TBD)

jcarrano, 2019-04-25:
Solved by #8847, #9153.

closed 2019-04-25 by jcarrano