RIOT-OS/RIOT #8436: Kinetis PhyNode: failed to flash binary > 256K

smlng, 2018-01-25:
#### Description

For release testing I tried to flash the (whole) unittest binary onto a PhyNode (`pba-d-01-kw2x`). First problem is, that the binary got quite large recently and exceeds 256K by now. Fortunately I got a PhyNode with Kinetis MCU marked as MKW22D512, which means it should have 512K ROM. Second problem: there are different versions of the PhyNode with differing ROM sizes (some have _only_ 256K)  , hence the default config assumes only 256K ROM size.

However, this can be overwritten by setting `CPU_MODEL` when compiling like this:

```
CPU_MODEL=mkw22d512vha5 KINETIS_INFO_DEBUG=1 BOARD=pba-d-01-kw2x make clean all flash term
```

Thanks to debug info output says:

```
Kinetis CPU info: M W 2 2 D _ 512 _ V HA 5
Qualification:     M
Core:              D
Series:            W
Family:            2
Subfamily:         2
ROM size:          512
RAM size:          64   (512/8)
SRAM_L:            32   (512/8/2)
Max speed:         5
Temperature range: V
Revision:          _
Package code:      HA
```

So we are all good? Nope, we are not, after successful compiling, flashing fails with:

```
### Flashing Target ###
/Volumes/devel/github/smlng/RIOT/tests/unittests/bin/pba-d-01-kw2x/tests_unittests.elf is fine.
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
Info : add flash_bank kinetis kw22d512.flash
adapter speed: 1000 kHz
cortex_m reset_config sysresetreq
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : CMSIS-DAP: FW Version = 1.0
Info : SWCLK/TCK = 0 SWDIO/TMS = 1 TDI = 0 TDO = 0 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 1000 kHz
Info : SWD DPIDR 0x2ba01477
START...
Info : kw22d512.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : MDM: Chip is unsecured. Continuing.
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* kw22d512.cpu       cortex_m   little kw22d512.cpu       running
START...
Info : MDM: Chip is unsecured. Continuing.
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000740 msp: 0x1fff8200
auto erase enabled
Info : Probing flash info for bank 0
Warn : Flash Configuration Field written.
Warn : Reset or power off the device to make settings effective.
Warn : no flash bank found for address 40000
Warn : no flash bank found for address 58e38
wrote 262144 bytes from file /Volumes/devel/github/smlng/RIOT/tests/unittests/bin/pba-d-01-kw2x/tests_unittests.elf in 14.380014s (17.802 KiB/s)
34 bytes written at address 0x20000000
downloaded 34 bytes in 0.003992s (8.317 KiB/s)
target halted due to breakpoint, current mode: Thread 
xPSR: 0x01000000 pc: 0x20000020 msp: 0x1fff8200
Error: checksum mismatch - attempting binary compare
[... 127 lines cut ...]
diff 127 address 0x0004007f. Was 0xff instead of 0x65
More than 128 errors, the rest are not printed.

make: *** [flash] Error 1
```

Looking at the output of openocd it seems that it does not recognise the additional flash banks (3 + 4) above 256K, and does not write beyond that, hence checksum fails and all addresses above 256K are FF.

#### Steps to reproduce the issue

Compile and (try to) flash `tests/unittests` onto board `pba-d-01-kw2x` (tweak CPU_MODEL as shown above).

#### Expected results

Flashing works.

#### Actual results

Flash error, see above

#### Versions

* RIOT 2018.01-RC1
* openocd Open On-Chip Debugger 0.10.0
* macOS 10.13.3

gebart, 2018-01-25:
I agree there's a problem with the flash detection, but AFAIK the chip does only have two banks, but they are twice as big as the 256k variant.
There have been some recent improvements to the openocd code regarding kinetis flash detection, it could be that this "just works" with a development version of openocd.
Another thing you can try is to use the frdm-common openocd.cfg. The pba-d-01-kw2x openocd.cfg is based on a sample configuration which was part of openocd v0.7, which is a bit outdated.

smlng, 2018-01-25:
@gebart thanks for the hints - I knew you might know something 😉. I'll a newer openocd and look into the frdm config.

kYc0o, 2018-07-19:
@smlng did you try?

smlng, 2018-07-19:
nope, not yet ... need to put on my list (again)

cgundogan, 2018-08-02:
@smlng might #9406 be a solution for this problem?

smlng, 2018-08-02:
@cgundogan nope doesn't help same error, seems openocd also only write the first (256K) flash bank but not to the second if the binary is larger

fjmolinas, 2019-04-19:
@smlng I think your issue is fix is you use a more recent version of openocd. The issue comes openocd script itself. If looking at master version:

https://github.com/ntfreak/openocd/blob/a4ac56152d9fc13c3fa479397407d9b86ffb13d8/tcl/target/kx.cfg#L40-L42

```
set _FLASHNAME $_CHIPNAME.pflash
flash bank $_FLASHNAME kinetis 0 0 0 0 $_TARGETNAME
kinetis create_banks

```
While looking at v0.10.0

https://github.com/ntfreak/openocd/blob/646566e006acd58fb9ac405511d3ecca79f26d55/tcl/target/kx.cfg#L37-L40

```
set _FLASHNAME $_CHIPNAME.flash
flash bank $_FLASHNAME kinetis 0 0 0 0 $_TARGETNAME
```

The extra `kinetis create_banks` command is not present in the stable openocd version. Could you compile openocd from source and verify this fixes your issue. If it does we might add some kind of warning comment like @cladmi suggested in #11174.

miri64, 2020-07-01:
Can you confirm @fjmolinas?

fjmolinas, 2020-07-03:
This is fixed in `openocd` since https://github.com/ntfreak/openocd/commit/c4d4c32a504f1a63f0200efdd175d21bfe8cc3af.

<details><summary>RIOT_TERMINAL=socat CPU_MODEL=mkw22d512vha5 BOARD=pba-d-01-kw2x make -C tests/unittests/ flash test --no-print-directory
</summary>

```
Building application "tests_unittests" for "pba-d-01-kw2x" with MCU "kinetis".

"make" -C /home/francisco/workspace/RIOT2/boards/pba-d-01-kw2x
"make" -C /home/francisco/workspace/RIOT2/core
"make" -C /home/francisco/workspace/RIOT2/cpu/kinetis
"make" -C /home/francisco/workspace/RIOT2/cpu/cortexm_common
"make" -C /home/francisco/workspace/RIOT2/cpu/cortexm_common/periph
"make" -C /home/francisco/workspace/RIOT2/cpu/kinetis/periph
"make" -C /home/francisco/workspace/RIOT2/drivers
"make" -C /home/francisco/workspace/RIOT2/drivers/mtd
"make" -C /home/francisco/workspace/RIOT2/drivers/periph_common
"make" -C /home/francisco/workspace/RIOT2/drivers/rtt_rtc
"make" -C /home/francisco/workspace/RIOT2/drivers/saul
"make" -C /home/francisco/workspace/RIOT2/drivers/sht1x
"make" -C /home/francisco/workspace/RIOT2/sys
"make" -C /home/francisco/workspace/RIOT2/sys/analog_util
"make" -C /home/francisco/workspace/RIOT2/sys/base64
"make" -C /home/francisco/workspace/RIOT2/sys/bitfield
"make" -C /home/francisco/workspace/RIOT2/sys/bloom
"make" -C /home/francisco/workspace/RIOT2/sys/checksum
"make" -C /home/francisco/workspace/RIOT2/sys/clif
"make" -C /home/francisco/workspace/RIOT2/sys/color
"make" -C /home/francisco/workspace/RIOT2/sys/crypto
"make" -C /home/francisco/workspace/RIOT2/sys/div
"make" -C /home/francisco/workspace/RIOT2/sys/ecc
"make" -C /home/francisco/workspace/RIOT2/sys/embunit
"make" -C /home/francisco/workspace/RIOT2/sys/event
"make" -C /home/francisco/workspace/RIOT2/sys/evtimer
"make" -C /home/francisco/workspace/RIOT2/sys/fmt
"make" -C /home/francisco/workspace/RIOT2/sys/frac
"make" -C /home/francisco/workspace/RIOT2/sys/fs/constfs
"make" -C /home/francisco/workspace/RIOT2/sys/hashes
"make" -C /home/francisco/workspace/RIOT2/sys/isrpipe
"make" -C /home/francisco/workspace/RIOT2/sys/luid
"make" -C /home/francisco/workspace/RIOT2/sys/matstat
"make" -C /home/francisco/workspace/RIOT2/sys/net/application_layer/gcoap
"make" -C /home/francisco/workspace/RIOT2/sys/net/application_layer/nanocoap
"make" -C /home/francisco/workspace/RIOT2/sys/net/ble/bluetil
"make" -C /home/francisco/workspace/RIOT2/sys/net/ble/bluetil/bluetil_addr
"make" -C /home/francisco/workspace/RIOT2/sys/net/credman
"make" -C /home/francisco/workspace/RIOT2/sys/net/crosslayer/inet_csum
"make" -C /home/francisco/workspace/RIOT2/sys/net/crosslayer/netopt
"make" -C /home/francisco/workspace/RIOT2/sys/net/gnrc
"make" -C /home/francisco/workspace/RIOT2/sys/net/gnrc/link_layer/mac
"make" -C /home/francisco/workspace/RIOT2/sys/net/gnrc/netapi
"make" -C /home/francisco/workspace/RIOT2/sys/net/gnrc/netif
"make" -C /home/francisco/workspace/RIOT2/sys/net/gnrc/netif/hdr
"make" -C /home/francisco/workspace/RIOT2/sys/net/gnrc/netreg
"make" -C /home/francisco/workspace/RIOT2/sys/net/gnrc/network_layer/icmpv6
"make" -C /home/francisco/workspace/RIOT2/sys/net/gnrc/network_layer/ipv6
"make" -C /home/francisco/workspace/RIOT2/sys/net/gnrc/network_layer/ipv6/hdr
"make" -C /home/francisco/workspace/RIOT2/sys/net/gnrc/network_layer/ipv6/nib
"make" -C /home/francisco/workspace/RIOT2/sys/net/gnrc/network_layer/ndp
"make" -C /home/francisco/workspace/RIOT2/sys/net/gnrc/network_layer/sixlowpan
"make" -C /home/francisco/workspace/RIOT2/sys/net/gnrc/network_layer/sixlowpan/ctx
"make" -C /home/francisco/workspace/RIOT2/sys/net/gnrc/network_layer/sixlowpan/frag/fb
"make" -C /home/francisco/workspace/RIOT2/sys/net/gnrc/network_layer/sixlowpan/frag/vrb
"make" -C /home/francisco/workspace/RIOT2/sys/net/gnrc/network_layer/sixlowpan/nd
"make" -C /home/francisco/workspace/RIOT2/sys/net/gnrc/pkt
"make" -C /home/francisco/workspace/RIOT2/sys/net/gnrc/pktbuf
"make" -C /home/francisco/workspace/RIOT2/sys/net/gnrc/pktbuf_static
"make" -C /home/francisco/workspace/RIOT2/sys/net/gnrc/priority_pktqueue
"make" -C /home/francisco/workspace/RIOT2/sys/net/gnrc/sock
"make" -C /home/francisco/workspace/RIOT2/sys/net/gnrc/sock/udp
[... 124 lines cut ...]
Info : Kinetis MK21DN512xxx5 detected: 2 flash blocks
Info : 2 PFlash banks: 512k total
auto erase enabled
wrote 358400 bytes from file /home/francisco/workspace/RIOT2/tests/unittests/bin/pba-d-01-kw2x/tests_unittests.elf in 12.334785s (28.375 KiB/s)

34 bytes written at address 0x20000000
downloaded 34 bytes in 0.003945s (8.417 KiB/s)

target halted due to breakpoint, current mode: Thread 
xPSR: 0x01000000 pc: 0x20000020 msp: 0x1fff8200
verified 357660 bytes in 2.267150s (154.060 KiB/s)

Info : MDM: Chip is unsecured. Continuing.
shutdown command invoked
Done flashing
r
socat - open:/dev/riot/tty-pba-d-01-kw2x,b115200,echo=0,raw
main(): This is RIOT! (Version: 2020.07-devel-1546-g49ab0-HEAD)
Help: Press s to start test, r to print it is ready
s
START
........................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................ [... cut ...] ......................................................................................................................................................................................................................................................................................................................................................................................................................................................................................
OK (998 tests)
```
</details>

closed 2020-07-03 by fjmolinas