RIOT-OS/RIOT #8446: periph/i2c: order of calling i2c_init_master and i2c_acquire

haukepetersen, 2018-01-26:
#### Description
Just saw something that might cause some trouble: currently we are calling `i2c_init_master` in all the device drivers that use I2C (yes thats broken, but this issue is not about that). We also call `i2c_acquire` in all the init functions. BUT the order in which we call these functions is not defined currently, so some drivers call `acquire` before `init_master`, others to it the other way around. The problem is now, that some i2c driver implemenations break this way, as they initialize their internal mutex in the `init_master` function, leading to undefined behavior when calling `mutex_lock` in `acquire()` before initializing the mutex. But then again, not acquiring the bus before calling `init_master` might also break things when more than one device uses the i2c bus, and their initialization is done in multithreaded contexts...

As I see it, there is no easy/quick fix for that in the current state of the I2C interface and this issue will be solved when moving to the reworked interface... But none the less, it exists and maybe this gives us some further pressure to finally get going on the i2c remodeling...

gebart, 2018-01-26:
Best short term solution IMO is to make sure the periph drivers use static initialization for the mutexes (`mutex_t m = MUTEX_INIT`), but that may be ugly when there are multiple busses.

haukepetersen, 2018-01-30:
Yes, I agree. Did a quick grep, and seems to me these are the only CPUs that need fixing:
- [ ] efm32
- [ ] nrf52

At least the nrf52 I broke, so I will have a quick take on fixing them :-)

haukepetersen, 2018-01-30:
@basilfx could you maybe have a quick look at the `efm32` i2c driver? Thx!

basilfx, 2018-01-30:
Should be done with #8489, I guess :-)

haukepetersen, 2018-03-02:
yes. All CPUs should be fixed for now. I tend to leave this issue open for future reference until the I2C remodeling is done, but feel free to close if anyone doesn't like this :)

cladmi, 2018-08-11:
@aabadie also found this while listing the release issues you may know more about it

kYc0o, 2018-08-15:
I'd tend to say that this is solved with the recently I2C merged work. Anyone can confirm an thus close the issue?

aabadie, 2018-10-17:
> Anyone can confirm and thus close the issue?

#10182 was a mistake from my side. I've checked all other I2C drivers and found nothing. Let's close this one.

closed 2018-10-17 by aabadie