RIOT-OS/RIOT #8458: release test 2018.01 RC1: tests/gnrc_ndp, kernel panic

A-Paul, 2018-01-26:
#### Description
Program crashes immediate after start.

#### Steps to reproduce the issue
"make clean all term" in *tests/gnrc_ndp*

#### Expected results

#### Actual results
> RIOT native interrupts/signals initialized.
> LED_RED_OFFRIOT native interrupts/signals initialized.
> LED_GREEN_ON
> RIOT native board initialized.
> RIOT native hardware initialization complete.
> 
> main(): This is RIOT! (Version: 2018.04-devel-17-ga78b-mobi31-2018.01-branch)
> .....................sys/net/gnrc/netif/gnrc_netif.c:1157 => 0x5663df32
> *** RIOT kernel panic:
> FAILED ASSERTION.
>
> *** halted.

#### Versions
```
----------------------------
          native gcc: gcc (Debian 6.3.0-18) 6.3.0 20170516
          msp430-gcc: msp430-gcc (GCC) 4.6.3 20120301 (mspgcc LTS 20120406 unpatched)
             avr-gcc: avr-gcc (GCC) 4.9.2
   arm-none-eabi-gcc: arm-none-eabi-gcc (15:5.4.1+svn241155-1) 5.4.1 20160919
    mips-mti-elf-gcc: missing/error
../../dist/tools/ci/print_toolchain_versions.sh: 92: ../../dist/tools/ci/print_toolchain_versions.sh: clang: not found
arm-none-eabi-newlib: "2.4.0"
 mips-mti-elf-newlib: missing/error
            avr-libc: "1.8.0svn" ("20111229")
            cppcheck: missing
          coccinelle: missing
                 git: git version 2.11.0
```

bergzand, 2018-01-26:
From what I've seen, the test network adapter used in the test doesn't implement `NETOPT_DEVICE_TYPE`, causing the assertion failure in gnrc_netif.

miri64, 2018-01-27:
Fix provided in #8464

PR #8464 (2018-01-27): tests/gnrc_ndp: add device type handler to netdev mock-up
### Contribution description
By providing a getter for `NETOPT_DEVICE_TYPE` the assertion failed in
#8458 is not hit anymore.

### Issues/PRs references
Fixes #8458
closed 2018-01-27 by smlng