RIOT-OS/RIOT #8473: SNTP not working

ilf-S, 2018-01-29:
#### Description
I'm trying to get NTP time from an SNTP server (Initially it was chrony, then openntpd). At first we tried with our own code in main.c, which was not successful, then switched to the sntp test code in /tests, but that doesn't work, too.

This is on a bluepill board with AT86RF212b transceivers. 6lowpan with RPi (where NTP daemon is) as BR. Tested with public servers, too, same result.

tcpdump shows an incoming NTPv4 packet from the bluepill and an outgoing one from the RPi. Compiled both SNTP and UDP with ENABLE_DEBUG(1), the output follows. All IPv6 addresses are public and routable. 
#### Steps to reproduce the issue
Compile the SNTP test on bluepill board with AT86RF212b transceiver (I suspect the result will be the same with any transceiver board combo). 
#### Expected results
Get ntpdate from server
#### Actual results
```
ntpdate 2001:XXX:XXXX:XXXX::1 100000
udp: GNRC_NETAPI_MSG_TYPE_SND
Error receiving message
Error in synchronization
> udp: GNRC_NETAPI_MSG_TYPE_RCV
udp: unable to forward packet as no one is interested in it
```
#### Versions
RIOT from trunk. main(): This is RIOT! (Version: 2018.04-devel-82-g0abbd-x220)
Compiled on Linux (Arch x86_64), gcc-arm-none-eabi-bin 7_2017_q4_major-1

tcpdump output:

```
20:40:40.641793 IP6 2001:XXX:XXXX:XXXX:7270:aa52:4269:ea76.49186 > 2001:XXX:XXXX:1044::1.123: NTPv4, Client, length 48

20:40:40.642868 IP6 2001:XXX:XXXX:1044::1.123 > 2001:XXX:XXXX:XXXX:7270:aa52:4269:ea76.49186: NTPv4, Server, length 48
```

miri64, 2018-01-29:
Can you maybe try to find out what the return value of the sync function is [here](https://github.com/RIOT-OS/RIOT/blob/master/sys/shell/commands/sc_sntp.c#L83-L86) and also what `errno` it corresponds to? I'll have a look at it myself tomorrow.

miri64, 2018-01-30:
Okay, I've tried it now myself. I wasn't able to do it with a Raspberry Pi, because mine doesn't seem to work at the moment so I found it easier to use the `gnrc_border_router` example instead with `openntpd` installed on my home machine. I also applied this patch:

```diff
diff --git a/sys/net/application_layer/sntp/sntp.c b/sys/net/application_layer/sntp/sntp.c
index 9d547fa..31f47a1 100644
--- a/sys/net/application_layer/sntp/sntp.c
+++ b/sys/net/application_layer/sntp/sntp.c
@@ -62,7 +62,8 @@ int sntp_sync(sock_udp_ep_t *server, uint32_t timeout)
                                      sizeof(_sntp_packet),
                                      timeout,
                                      NULL)) < 0) {
-        DEBUG("Error receiving message\n");
+        errno = -result;
+        perror("Error receiving message");
         sock_udp_close(&_sntp_sock);
         return result;
     }
```

With this, when I tried with the `100000` you proposed I got an error as well (Connection timed out), but with `1000000` I got a result. I'll try to find a better default timeout and set it in a PR accordingly.

miri64, 2018-01-30:
> I'll try to find a better default timeout and set it in a PR accordingly.

See #8478

ilf-S, 2018-01-30:
Damn, timeout of 1000000 indeed does work for me, too.

```
> ntpdate 2001:XXX:XXXX:XXXX::1 1000000
udp: GNRC_NETAPI_MSG_TYPE_SND
udp: GNRC_NETAPI_MSG_TYPE_RCV
2018-01-30 15:41:35 UTC (-515257269 us)
> 
```

We will need to adjust our code, too. Too bad the RTC does not seem to work on STM32F1s, but we will move to F4s anyway. Thank you for the help. I thought it was an UDP issue at the begining and it was a legitimate issue. Should have send an email to the mailing list first. Should I close this?

miri64, 2018-01-30:
Should be closed, when #8478 is merged, so leave it open, please.

PR #8478 (2018-01-30): shell_commands: sntp: increase default timeout
### Contribution description
This PR

* specifies the time unit used for the timeout with the `ntpdate` command
* increases the default timeout as 5000μs is far to small

### Issues/PRs references
Fixes #8473
closed 2018-01-30 by cladmi