RIOT-OS/RIOT #8477: tests: gnrc_ipv6_nib_6ln fails

smlng, 2018-01-30:
#### Description

Test application `tests/gnrc_ipv6_nib_6ln` fails with 4 errors, on macOS *and* Linux.

#### Steps to reproduce the issue

run `make tests` for named tests app

#### Expected results

success without error

#### Actual results

4 errors:

```
/RIOT/tests/gnrc_ipv6_nib_6ln/bin/native/tests_gnrc_ipv6_nib_6ln.elf   
RIOT native interrupts/signals initialized.
LED_RED_OFF
LED_GREEN_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

main(): This is RIOT! (Version: 2018.04-devel-90-gb138a-mobi25.inet.haw-hamburg.de-fix_tests_180129)
.........................................
tests.test_handle_pkt__rtr_adv__success_pio_L0 (tests/gnrc_ipv6_nib_6ln/main.c 1075) _PIO_PFX_LTIME < prefix.valid_until
..
tests.test_handle_pkt__rtr_adv__success_pio_LA (tests/gnrc_ipv6_nib_6ln/main.c 1075) _PIO_PFX_LTIME < prefix.valid_until
...
tests.test_handle_pkt__rtr_adv__success_6co_sl2ao_mtuo_pio_L0 (tests/gnrc_ipv6_nib_6ln/main.c 1075) _PIO_PFX_LTIME < prefix.valid_until
..
tests.test_handle_pkt__rtr_adv__success_6co_sl2ao_mtuo_pio_LA (tests/gnrc_ipv6_nib_6ln/main.c 1075) _PIO_PFX_LTIME < prefix.valid_until
.
run 49 failures 4
Timeout in expect script at "child.expect(r"OK \(\d+ tests\)")" (tests/gnrc_ipv6_nib_6ln/tests/01-run.py:15)

make: *** [test] Error 1
```

#### Versions

OS: macOS 10.13.03 with

```
Installed toolchain versions
----------------------------
          native gcc: gcc (Homebrew GCC 7.2.0_1) 7.2.0
          msp430-gcc: missing/error
             avr-gcc: avr-gcc (GCC) 7.3.0
   arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
    mips-mti-elf-gcc: missing/error
               clang: Apple LLVM version 9.0.0 (clang-900.0.39.2)
arm-none-eabi-newlib: "2.5.0"
 mips-mti-elf-newlib: missing/error
            avr-libc: "2.0.0" ("20150208")
            cppcheck: Cppcheck 1.82
          coccinelle: spatch version 1.0.6 compiled with OCaml version 4.05.0
                 git: git version 2.16.1
```

Linux: CentOS 7 with

```
Installed toolchain versions
----------------------------
          native gcc: gcc (GCC) 6.3.1 20170216 (Red Hat 6.3.1-3)
          msp430-gcc: missing/error
             avr-gcc: missing/error
   arm-none-eabi-gcc: missing/error
    mips-mti-elf-gcc: missing/error
               clang: clang version 3.4.2 (tags/RELEASE_34/dot2-final)
arm-none-eabi-newlib: missing/error
 mips-mti-elf-newlib: missing/error
            avr-libc: missing/error (missing/error)
            cppcheck: missing
          coccinelle: missing
                 git: git version 1.8.3.1
```

miri64, 2018-01-30:
(`tests/gnrc_ipv6_nib` isn't working either, right?)

miri64, 2018-01-30:
This was broken in https://github.com/RIOT-OS/RIOT/pull/8135. Will provide a fix.

smlng, 2018-01-30:
> (tests/gnrc_ipv6_nib isn't working either, right?)

well, yes error messages ...

miri64, 2018-01-30:
See #8480 (another reason why testing corner cases is good.... though we should have run the tests in #8135 ;-)

PR #8480 (2018-01-30): gnrc_ipv6_nib: ignore corner case when adding to PL
### Contribution description
In #8135 the handling of corner cases for the conversion of
milliseconds to seconds, but the internal handling was not adapted.

This fixes that and also the tests for the NIB.

### Issues/PRs references
Fixes #8477
closed 2018-01-30 by miri64