RIOT-OS/RIOT #8498: gnrc: gnrc_networking: problem picking up public address from RasPi + openlab

kaspar030, 2018-01-31:
#### Description

I have a fairly standard Raspberry PI + Openlab that's announcing a /64 prefix to the 802.15.4 network.
With 2017.10, this works fine. Current master doesn't pick up the addresses.

#### Steps to reproduce the issue

1. follow guides to set up openlab & radvd
2. flash gnrc_networking to samr21-xpro

#### Expected results

ifconfig shows public address

#### Actual results

ifconfig doesn't show public address

#### Versions

raspian stretch, "Linux pi-0923cd37 4.9.41-v7+", iwpan version 0.7

jia200x, 2018-01-31:
@fjmolinas and @paula75 were working with a similar setup, and I think they weere able to pick up addresses. Is it still working?

smlng, 2018-01-31:
I cannot confirm the issue either, tested with RasPi with OpenLabs and `gnrc_networking` on PhyNode (pba-d-01-kw2x) and samr21-xpro: both get a public address though it took the samr21 a bit longer than the PhyNode.

Output samr21-xpro:

```
> ifconfig
2018-01-31 13:55:37,492 - INFO #  ifconfig
2018-01-31 13:55:37,498 - INFO # Iface  7  HWaddr: 51:be  Channel: 18  Page: 0  NID: 0x2424
2018-01-31 13:55:37,502 - INFO #           Long HWaddr: 79:62:34:69:7b:bd:d1:be 
2018-01-31 13:55:37,509 - INFO #            TX-Power: 0dBm  State: IDLE  max. Retrans.: 3  CSMA Retries: 4 
2018-01-31 13:55:37,515 - INFO #           AUTOACK  ACK_REQ  CSMA  MTU:1280  HL:64  RTR  
2018-01-31 13:55:37,517 - INFO #           IPHC  
2018-01-31 13:55:37,520 - INFO #           Source address length: 8
2018-01-31 13:55:37,522 - INFO #           Link type: wireless
2018-01-31 13:55:37,529 - INFO #           inet6 addr: fe80::7b62:3469:7bbd:d1be  scope: local  VAL
2018-01-31 13:55:37,531 - INFO #           inet6 group: ff02::2
2018-01-31 13:55:37,534 - INFO #           inet6 group: ff02::1
2018-01-31 13:55:37,539 - INFO #           inet6 group: ff02::1:ffbd:d1be
2018-01-31 13:55:37,541 - INFO #           inet6 group: ff02::1a
2018-01-31 13:55:37,542 - INFO #           
2018-01-31 13:55:37,545 - INFO #           Statistics for Layer 2
2018-01-31 13:55:37,548 - INFO #             RX packets 0  bytes 0
2018-01-31 13:55:37,552 - INFO #             TX packets 8 (Multicast: 8)  bytes 461
2018-01-31 13:55:37,556 - INFO #             TX succeeded 8 errors 0
2018-01-31 13:55:37,558 - INFO #           Statistics for IPv6
2018-01-31 13:55:37,561 - INFO #             RX packets 0  bytes 0
2018-01-31 13:55:37,566 - INFO #             TX packets 8 (Multicast: 8)  bytes 498
2018-01-31 13:55:37,569 - INFO #             TX succeeded 8 errors 0
2018-01-31 13:55:37,569 - INFO # 
> ifconfig
2018-01-31 14:00:10,366 - INFO #  ifconfig
2018-01-31 14:00:10,372 - INFO # Iface  7  HWaddr: 51:be  Channel: 18  Page: 0  NID: 0x2424
2018-01-31 14:00:10,376 - INFO #           Long HWaddr: 79:62:34:69:7b:bd:d1:be 
2018-01-31 14:00:10,383 - INFO #            TX-Power: 0dBm  State: IDLE  max. Retrans.: 3  CSMA Retries: 4 
2018-01-31 14:00:10,389 - INFO #           AUTOACK  ACK_REQ  CSMA  MTU:1280  HL:255  RTR  
2018-01-31 14:00:10,392 - INFO #           RTR_ADV  IPHC  
2018-01-31 14:00:10,395 - INFO #           Source address length: 8
2018-01-31 14:00:10,398 - INFO #           Link type: wireless
2018-01-31 14:00:10,404 - INFO #           inet6 addr: fe80::7b62:3469:7bbd:d1be  scope: local  VAL
2018-01-31 14:00:10,413 - INFO #           inet6 addr: fd16:abcd:ef24:3:7b62:3469:7bbd:d1be  scope: global  VAL
2018-01-31 14:00:10,414 - INFO #           inet6 group: ff02::2
2018-01-31 14:00:10,416 - INFO #           inet6 group: ff02::1
2018-01-31 14:00:10,420 - INFO #           inet6 group: ff02::1:ffbd:d1be
2018-01-31 14:00:10,423 - INFO #           inet6 group: ff02::1a
2018-01-31 14:00:10,424 - INFO #           
2018-01-31 14:00:10,427 - INFO #           Statistics for Layer 2
2018-01-31 14:00:10,430 - INFO #             RX packets 16  bytes 1155
2018-01-31 14:00:10,435 - INFO #             TX packets 29 (Multicast: 9)  bytes 2782
2018-01-31 14:00:10,438 - INFO #             TX succeeded 29 errors 0
2018-01-31 14:00:10,441 - INFO #           Statistics for IPv6
2018-01-31 14:00:10,444 - INFO #             RX packets 16  bytes 1248
2018-01-31 14:00:10,449 - INFO #             TX packets 29 (Multicast: 9)  bytes 2162
2018-01-31 14:00:10,452 - INFO #             TX succeeded 29 errors 0
2018-01-31 14:00:10,453 - INFO # 
```

output PhyNode:
```
2018-01-31 13:29:16,286 - INFO # main(): This is RIOT! (Version: 2018.01-mobi25.inet.haw-hamburg.de-2018.01-branch)
2018-01-31 13:29:16,290 - INFO # RIOT network stack example application
2018-01-31 13:29:16,292 - INFO # All up, running the shell now
> ifconfig
2018-01-31 13:29:23,339 - INFO #  ifconfig
2018-01-31 13:29:23,343 - INFO # Iface  7  HWaddr: 13:36  Channel: 26  NID: 0x23
2018-01-31 13:29:23,347 - INFO #           Long HWaddr: d3:c1:6d:73:ab:43:13:36 
2018-01-31 13:29:23,353 - INFO #            TX-Power: 0dBm  State: IDLE 
2018-01-31 13:29:23,354 - INFO #           ACK_REQ  AUTOCCA
2018-01-31 13:29:23,357 - INFO #           MTU:1280  HL:64  RTR  IPHC  
2018-01-31 13:29:23,360 - INFO #           Source address length: 8
2018-01-31 13:29:23,363 - INFO #           Link type: wireless
[... 24 lines cut ...]
2018-01-31 13:29:37,927 - INFO #           Long HWaddr: d3:c1:6d:73:ab:43:13:36 
2018-01-31 13:29:37,933 - INFO #            TX-Power: 0dBm  State: IDLE 
2018-01-31 13:29:37,934 - INFO #           ACK_REQ  AUTOCCA
2018-01-31 13:29:37,938 - INFO #           MTU:1280  HL:255  RTR  RTR_ADV  IPHC  
2018-01-31 13:29:37,941 - INFO #           Source address length: 8
2018-01-31 13:29:37,943 - INFO #           Link type: wireless
2018-01-31 13:29:37,949 - INFO #           inet6 addr: fe80::d1c1:6d73:ab43:1336  scope: local  VAL
2018-01-31 13:29:37,956 - INFO #           inet6 addr: fd16:abcd:ef24:3:d1c1:6d73:ab43:1336  scope: global  VAL
2018-01-31 13:29:37,959 - INFO #           inet6 group: ff02::2
2018-01-31 13:29:37,961 - INFO #           inet6 group: ff02::1
2018-01-31 13:29:37,966 - INFO #           inet6 group: ff02::1:ff43:1336
2018-01-31 13:29:37,968 - INFO #           inet6 group: ff02::1a
2018-01-31 13:29:37,969 - INFO #           
2018-01-31 13:29:37,972 - INFO #           Statistics for Layer 2
[... cut ...]
2018-01-31 13:29:37,979 - INFO #             TX packets 7 (Multicast: 5)  bytes 377
2018-01-31 13:29:37,982 - INFO #             TX succeeded 6 errors 0
2018-01-31 13:29:37,985 - INFO #           Statistics for IPv6
2018-01-31 13:29:37,988 - INFO #             RX packets 1  bytes 128
2018-01-31 13:29:37,992 - INFO #             TX packets 7 (Multicast: 5)  bytes 498
2018-01-31 13:29:37,996 - INFO #             TX succeeded 7 errors 0
2018-01-31 13:29:37,996 - INFO # 
```

I'm using release tag `2018.01`

kaspar030, 2018-01-31:
On my setup it output this (2018.01):

```
> ifconfig
2018-01-31 14:04:50,898 - INFO #  ifconfig
2018-01-31 14:04:50,904 - INFO # Iface  7  HWaddr: 5b:ce  Channel: 26  Page: 0  NID: 0x23
2018-01-31 14:04:50,908 - INFO #           Long HWaddr: 79:76:62:60:b7:e3:db:ce
2018-01-31 14:04:50,915 - INFO #            TX-Power: 0dBm  State: IDLE  max. Retrans.: 3  CSMA Retries: 4
2018-01-31 14:04:50,921 - INFO #           AUTOACK  ACK_REQ  CSMA  MTU:1280  HL:64  RTR
2018-01-31 14:04:50,923 - INFO #           IPHC
2018-01-31 14:04:50,926 - INFO #           Source address length: 8
2018-01-31 14:04:50,929 - INFO #           Link type: wireless
2018-01-31 14:04:50,935 - INFO #           inet6 addr: fe80::7b76:6260:b7e3:dbce  scope: local  VAL
2018-01-31 14:04:50,938 - INFO #           inet6 group: ff02::2
2018-01-31 14:04:50,940 - INFO #           inet6 group: ff02::1
2018-01-31 14:04:50,944 - INFO #           inet6 group: ff02::1:ffe3:dbce
2018-01-31 14:04:50,947 - INFO #           inet6 group: ff02::1a
2018-01-31 14:04:50,948 - INFO #
2018-01-31 14:04:50,951 - INFO #           Statistics for Layer 2
2018-01-31 14:04:50,954 - INFO #             RX packets 2  bytes 172
2018-01-31 14:04:50,959 - INFO #             TX packets 4 (Multicast: 4)  bytes 217
2018-01-31 14:04:50,962 - INFO #             TX succeeded 4 errors 0
2018-01-31 14:04:50,965 - INFO #           Statistics for IPv6
2018-01-31 14:04:50,968 - INFO #             RX packets 2  bytes 208
2018-01-31 14:04:50,973 - INFO #             TX packets 4 (Multicast: 4)  bytes 242
2018-01-31 14:04:50,976 - INFO #             TX succeeded 4 errors 0
2018-01-31 14:04:50,976 - INFO #
>
```

kaspar030, 2018-01-31:
Bisecting is difficult, newest known-good merge on master is ```b1e69a776fbdc546557c242509654d6292ea1cdf```, I'm trying to find a newer one.

Here it outputs:

```
> ifconfig
2018-01-31 14:05:55,940 - INFO #  ifconfig
2018-01-31 14:05:55,945 - INFO # Iface  7   HWaddr: 5b:ce  Channel: 26  Page: 0  NID: 0x23
2018-01-31 14:05:55,950 - INFO #            Long HWaddr: 79:76:62:60:b7:e3:db:ce
2018-01-31 14:05:55,957 - INFO #            TX-Power: 0dBm  State: IDLE  max. Retrans.: 3  CSMA Retries: 4
2018-01-31 14:05:55,963 - INFO #            ACK_REQ  CSMA  MTU:1280  HL:64  6LO  RTR  RTR_ADV  IPHC
2018-01-31 14:05:55,967 - INFO #            Source address length: 8
2018-01-31 14:05:55,969 - INFO #            Link type: wireless
2018-01-31 14:05:55,975 - INFO #            inet6 addr: ff02::1/128  scope: local [multicast]
2018-01-31 14:05:55,981 - INFO #            inet6 addr: fe80::7b76:6260:b7e3:dbce/64  scope: local
2018-01-31 14:05:55,987 - INFO #            inet6 addr: ff02::1:ffe3:dbce/128  scope: local [multicast]
2018-01-31 14:05:55,993 - INFO #            inet6 addr: ff02::1a/128  scope: local [multicast]
2018-01-31 14:05:56,000 - INFO #            inet6 addr: 2001:470:749d:4:7b76:6260:b7e3:dbce/64  scope: global
2018-01-31 14:05:56,005 - INFO #            inet6 addr: ff02::2/128  scope: local [multicast]
2018-01-31 14:05:56,006 - INFO #
2018-01-31 14:05:56,009 - INFO #            Statistics for Layer 2
2018-01-31 14:05:56,013 - INFO #             RX packets 4  bytes 278
2018-01-31 14:05:56,017 - INFO #             TX packets 6 (Multicast: 2)  bytes 547
2018-01-31 14:05:56,020 - INFO #             TX succeeded 6 errors 0
2018-01-31 14:05:56,023 - INFO #            Statistics for IPv6
2018-01-31 14:05:56,026 - INFO #             RX packets 4  bytes 328
2018-01-31 14:05:56,031 - INFO #             TX packets 6 (Multicast: 2)  bytes 434
2018-01-31 14:05:56,034 - INFO #             TX succeeded 6 errors 0
2018-01-31 14:05:56,034 - INFO #
>
```

kaspar030, 2018-01-31:
radvd "Version: 2.15"

radvd.conf:

```
root@pi-0923cd37:~# cat /etc/radvd.conf
interface lowpan0 {
        AdvSendAdvert on;
        MinRtrAdvInterval 2;
        MaxRtrAdvInterval 10;

        prefix  2001:470:749d:4::1/64 {
                AdvOnLink off;
                AdvAutonomous on;
                AdvRouterAddr on;
        };
};
```

kaspar030, 2018-01-31:
Any information that might help?

smlng, 2018-01-31:
my config:

```
radvd -v
Version: 2.13

Compiled in settings:
  default config file		"/etc/radvd.conf"
  default pidfile		"/var/run/radvd.pid"
  default logfile		"/var/log/radvd.log"
  default syslog facility	24
Please send bug reports or suggestions to Reuben Hawkins <reubenhwk@gmail.com>.
pi@IoT24:~ $ cat /etc/radvd.conf 
interface lowpan0
{
    AdvSendAdvert on;
    UnicastOnly on;
    AdvCurHopLimit 255;
    AdvSourceLLAddress on;

    prefix fd16:abcd:ef24:3::/64
    {
        AdvOnLink off;
        AdvAutonomous on;
        AdvRouterAddr on;
    };

    abro fe80::1ac0:ffee:c0ff:ee24/64
    {
        AdvVersionLow 10;
        AdvVersionHigh 2;
        AdvValidLifeTime 2;
    };
};
```

smlng, 2018-01-31:
maybe you need an ABRO now, with new ND merged?

kaspar030, 2018-01-31:
> maybe you need an ABRO now, with new ND merged?

How do I set that?

smlng, 2018-01-31:
or `UnicastOnly on;`

ABRO: see my config above

smlng, 2018-01-31:
must be the link-local address of the pi lowpan interface.

kaspar030, 2018-01-31:
> maybe you need an ABRO now, with new ND merged?

Thats was it!

kaspar030, 2018-01-31:
Alright, thanks for helping, I guess we don't have a showstopper.

kaspar030, 2018-01-31:
Might want to add a note to the release notes...

closed 2018-01-31 by kaspar030

miri64, 2018-01-31:
> Might want to add a note to the release notes...

"NDP now RFC-compliant" ^^. Just for the other maintainers, @kaspar030 and I discussed this. The reason why ABRO is required is because of this sentence 

> If a received RA does not contain an ABRO, then the RA MUST be silently ignored.

In [RFC 6775, section 8.1.3](https://tools.ietf.org/html/rfc6775#section-8.1.3). The other way to fix this issue (e.g. if changing the RADVD is not possible for what ever reason) is to deactivate Multihop Prefix and Context Distribution (by setting `GNRC_IPV6_NIB_CONF_MULTIHOP_P6C=0` at compile time). Than prefixes will only be distributed single hop and no compression contexts are distributed either.