RIOT-OS/RIOT #8510: UART_STDIO_BAUDRATE and BAUD merger

Josar, 2018-02-02:
The situation right now is that both UART_STDIO_BAUDRATE and BAUD have to be defined. But in my understanding they are always the same. 

But havig to change two settings, on multiple places, to be able to compile and run the board with different baud seem not very intuitiv and error prone to me.

So my suggestion would be:

1. Implement ifdefs for all boards for UART_STDIO_BAUDRATE, then it can be set in the makefile.

1. Then instead of defining BAUD, rather defining UART_STDIO_BAUDRATE as export in the board/Makefile.include and also use UART_STDIO_BAUDRATE instead of BAUD for the pyterm.

1. And Add the cflag ??? i don't know where. I'm not a fan of distrributing everything so i would say in the board/makefile.include.

Best regards
Josua

PeterKietzmann, 2018-02-19:
I never really thought about it but your point seems reasonable. I personally have not real preference. Maybe @haukepetersen has?

Josar, 2018-03-09:
Anyone some thoughts about this?

Josar, 2018-04-27:
Still open for discussion.

closed 2019-09-10 by stale[bot]