RIOT-OS/RIOT #8523: tools: need backport script

kaspar030, 2018-02-06:
#### Description

Creating backports should be automated to something like

```
$ dist/tools/release/backport.sh #prnum
```

... which would automatically create a new branch on top of the last release branch, using all commits from #prnum, and create a PR (possibly using "hub"), automatically filling subject and description.

IMO that could save a lot of time...

miri64, 2018-02-06:
Already on my TODO list. I take the assignment for now, but if somebody else is faster, I'm not mad ;-) (but somebody knowledgeable with git should do this, as IMHO finding the right commits to rebase onto the release branch is not a simple task IMHO, especially if the PR is already merged into master).

PeterKietzmann, 2019-02-21:
#8968 was merged recently

closed 2019-02-21 by PeterKietzmann