RIOT-OS/RIOT #8548: arduino-uno does not provide FEATURES_PROVIDED=cpp

gebart, 2018-02-12:
#### Description
C++ support is required to build Arduino code. Building the arduino-hello-world example with BOARD=arduino-uno works just fine, but the build system is warning that feature cpp is missing for the board.

#### Steps to reproduce the issue
```
cd examples/arduino_hello-world
make BOARD=arduino-uno
```

#### Expected results
No warnings

#### Actual results
```
There are unsatisfied feature requirements: cpp

EXPECT ERRORS!
```
but compilation succeeds (using both BUILD_IN_DOCKER=1 and my local avr-gcc installation)

#### Versions
riotbuild docker image b329486bb877

avr-gcc: avr-gcc (Gentoo 7.2.0-r1 p1.1) 7.2.0
avr-libc: "2.0.0" ("20150208")

kaspar030, 2018-04-16:
Hm, just adding ```cpp``` to the CPU's features won't work, as on avr there's no libstdc++, which all the other cpp tests require...

kaspar030, 2018-04-17:
How about removing the feature requirement for certain whitelisted boards?

gebart, 2018-04-17:
I propose the Arduino build layer use FEATURES_REQUIRED += arduino instead of cpp

gebart, 2018-04-17:
Btw, what exactly is missing in an Arduino toolchain? It can't be libstdc++, because that is part of gcc, not libc

kaspar030, 2018-04-20:
> I propose the Arduino build layer use FEATURES_REQUIRED += arduino instead of cpp

+1

> Btw, what exactly is missing in an Arduino toolchain? It can't be libstdc++, because that is part of gcc

I don't think there is an libstdc++ implementation for AVR that is freely available.

cladmi, 2018-04-20:
No error message anymore when doing

```make BOARD=arduino-uno -C examples/arduino_hello-world/```

PR #8995 (2018-04-20): sys/arduino: drop cpp feature dependency
### Contribution description

Currently, the arduino module depends on the "cpp" feature. As AVR's don't have ```libstdc++```, they don't provide that feature, as otherwise all cpp examples would break.
But the AVR does provide a cpp compiler and thus the arduino module compiles just fine.

This PR removes the cpp dependency from arduino module.

### Issues/PRs references

Fixes #8548.
closed 2018-04-20 by cladmi