RIOT-OS/RIOT #8550: tests/unittests fail on OSX

kYc0o, 2018-02-13:
#### Description

Unittests fail for OSX

#### Steps to reproduce the issue

`make -C tests/unittests all test`

#### Expected results

All unittests pass.

#### Actual results

```console
RIOT native interrupts/signals initialized.
LED_RED_OFF
LED_GREEN_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

main(): This is RIOT! (Version: 2018.07-devel-155-g6f46b-snake.local)
...................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................lfs error: Corrupted dir pair at 0 1
lfs error: Invalid superblock at -1549646892 1056804
lfs error: Corrupted dir pair at 0 1
lfs error: Invalid superblock at -1549646892 1056804
................................................................................................................................................................................................Unexpected end of file in expect script at "child.expect(u"OK \\([0-9]+ tests\\)")" (tests/unittests/tests/01-run.py:14)

Process already stopped
make: *** [Makefile:236: test] Error 1
make: Leaving directory '/Users/facosta/git/RIOT-OS/RIOT/tests/unittests'
```

#### Versions

```console
Installed toolchain versions
----------------------------
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
          native gcc: Apple LLVM version 8.0.0 (clang-800.0.42.1)
          msp430-gcc: msp430-gcc (GCC) 4.7.0 20120322 (mspgcc dev 20120911)
             avr-gcc: avr-gcc (GCC) 7.3.0
   arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
    mips-mti-elf-gcc: missing/error
               clang: Apple LLVM version 8.0.0 (clang-800.0.42.1)
arm-none-eabi-newlib: "2.5.0"
 mips-mti-elf-newlib: missing/error
            avr-libc: "2.0.0" ("20150208")
            cppcheck: missing
          coccinelle: missing
                 git: git version 2.16.1
```

kaspar030, 2018-04-11:
still?

kYc0o, 2018-04-11:
It's not really clear:

```console
RIOT native interrupts/signals initialized.
LED_RED_OFF
LED_GREEN_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

main(): This is RIOT! (Version: 2018.07-devel-862-g578fb-snake.local)
..........................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................lfs error:469: Corrupted dir pair at 0 1
lfs error:2202: Invalid superblock at 0 1
lfs error:469: Corrupted dir pair at 0 1
lfs error:2202: Invalid superblock at 0 1
........................................................................................................................................................................................................make: Leaving directory '/Users/facosta/git/RIOT-OS/RIOT/tests/unittests'
```

on Linux
```console
RIOT native interrupts/signals initialized.
LED_RED_OFF
LED_GREEN_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

main(): This is RIOT! (Version: 2018.04-devel-928-g578fb-vagrant)
.........................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................lfs error:469: Corrupted dir pair at 0 1
lfs error:2202: Invalid superblock at 0 1
lfs error:469: Corrupted dir pair at 0 1
lfs error:2202: Invalid superblock at 0 1
.................................................................................................................................................................
OK (890 tests)
```

but I guess everything runs OK.

vincent-d, 2018-04-12:
Not sure if it's related, but #8934 disables the littlefs prints. Might be worth to try.

kYc0o, 2018-04-12:
After #8934 was merged I don't get anymore the errors, but for some reason the terminal is killed before the "OK" is printed. Though it doesn't mean the unittests failed it's a problem. Anyways I think we can close this.

```console
RIOT native interrupts/signals initialized.
LED_RED_OFF
LED_GREEN_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

main(): This is RIOT! (Version: 2018.07-devel-892-g3be4b-snake.local)
..................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................make: Leaving directory '/Users/facosta/git/RIOT-OS/RIOT/tests/unittests'
```

kaspar030, 2018-04-16:
Is the test succeeding? (```make test```)

kYc0o, 2018-04-16:
Nope, because for some reason the make recipe ends before the `OK (890 tests)` is printed. I need to investigate if it's related to the recent make modifications.

kYc0o, 2018-04-16:
For info, in current master:

```console
RIOT native interrupts/signals initialized.
LED_RED_OFF
LED_GREEN_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

main(): This is RIOT! (Version: 2018.07-devel-945-gfbea4-snake.local)
...................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................Unexpected end of file in expect script at "child.expect(u"OK \\([0-9]+ tests\\)")" (tests/unittests/tests/01-run.py:14)
```

kYc0o, 2018-04-16:
hmmm... it doesn't seem to be a general issue, but maybe only on my computer. @smlng can you test this to compare?

smlng, 2018-04-16:
well, with latest master I get the following interesting error:

```
git clone https://github.com/RIOT-OS/RIOT
Cloning into 'RIOT'...
remote: Counting objects: 139539, done.
remote: Compressing objects: 100% (19/19), done.
remote: Total 139539 (delta 4), reused 16 (delta 3), pack-reused 139517
Receiving objects: 100% (139539/139539), 49.76 MiB | 5.66 MiB/s, done.
Resolving deltas: 100% (88255/88255), done.
Checking out files: 100% (5488/5488), done.
iMac:tmp sebastian$ cd RIOT/tests/unittests/
iMac:unittests sebastian$ make all
Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Building application "tests_unittests" for "native" with MCU "native".

make: *** No rule to make target `/private/tmp/RIOT/tests/unittests/bin/native/tests-base64.a', needed by `/private/tmp/RIOT/tests/unittests/bin/native/tests_unittests.elf'.  Stop.
```

kYc0o, 2018-04-16:
Yes that's a known issue not only for OS X. Can you try with `make clean all term`?

smlng, 2018-04-16:
ah yes, read and remember the issue ... with `make clean all term` I get another error:

```
make clean all term
Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
make: Nothing to be done for `clean'.
Building application "tests_unittests" for "native" with MCU "native".

make[1]: Nothing to be done for `prepare'.
make[1]: Nothing to be done for `prepare'.
make[1]: Nothing to be done for `prepare'.
make[1]: Nothing to be done for `prepare'.
make[1]: Nothing to be done for `prepare'.
make[1]: Nothing to be done for `prepare'.
make[1]: Nothing to be done for `prepare'.
make[1]: Nothing to be done for `prepare'.
make[1]: Nothing to be done for `prepare'.
"/Library/Developer/CommandLineTools/usr/bin/make" -C /private/tmp/RIOT/pkg/cn-cbor
"/Library/Developer/CommandLineTools/usr/bin/make" -C /private/tmp/RIOT/tests/unittests/bin/pkg/native/cn-cbor/src -f /private/tmp/RIOT/pkg/cn-cbor/Makefile.cn-cbor
"/Library/Developer/CommandLineTools/usr/bin/make" -C /private/tmp/RIOT/pkg/hacl
"/Library/Developer/CommandLineTools/usr/bin/make" -C /private/tmp/RIOT/tests/unittests/bin/pkg/native/hacl -f /private/tmp/RIOT/pkg/hacl/Makefile.hacl
"/Library/Developer/CommandLineTools/usr/bin/make" -C /private/tmp/RIOT/pkg/heatshrink
"/Library/Developer/CommandLineTools/usr/bin/make" -C /private/tmp/RIOT/tests/unittests/bin/pkg/native/heatshrink -f /private/tmp/RIOT/pkg/heatshrink/Makefile.heatshrink
"/Library/Developer/CommandLineTools/usr/bin/make" -C /private/tmp/RIOT/pkg/libcose
"/Library/Developer/CommandLineTools/usr/bin/make" -C /private/tmp/RIOT/tests/unittests/bin/pkg/native/libcose/src -f /private/tmp/RIOT/pkg/libcose/Makefile.libcose
make: *** /private/tmp/RIOT/tests/unittests/bin/pkg/native/libcose/src: No such file or directory.  Stop.
make[1]: *** [all] Error 2
make: *** [/private/tmp/RIOT/tests/unittests/bin/native/libcose.a] Error 2
```

kYc0o, 2018-07-20:
I could run all the unittests successfully on OS X. However, for some reason the target `test` throws a timeout, but that's another issue. Thus, closing.

closed 2018-07-20 by kYc0o