RIOT-OS/RIOT #8589: Why using -F in avrdude?

Citrullin, 2018-02-18:
Hello,

I got a simple question. Why do you force avrdude?
I configured a ATMega328P. So I used the arduino Uno configuration. I use a USBASP, so I set the port to /dev/USBasp. I executed make flash. Which resulted in a broken board, because riot uses -p arduino instead of usbasp. I don't know what exactly is broken now. But I though: "Why using the -F flag in avrdude for normal programming? That's a really hard way to flash it to the MCU. "

Since the board cost around 2 Euro I don't invest more time of finding the issue. I just use a new one.

roberthartung, 2018-02-22:
Hi,
good question. I can't really tell you and the one who wrote this line should be able to tell us. #6516 Was the PR for that and @antmerlino is the author. Maybe he can tell us?
I don't see a particular reason. Never needed to use `-F` myself.

- Robert

miri64, 2020-07-01:
From what I can tell, the `-F` flag is not used anymore with `avrdude`, at least I can't find it in any of the usual places and there is no alias for it according to the `manpage`.

```
$ git grep -e "-F\>" -- cpu boards/ makefiles/
Binary file cpu/esp32/bin/rom_0x40000000_0x000c2000.bin matches
cpu/kinetis/dist/check-fcfield.sh:    get_fc_field "${1}" | awk -F' ' "/${2}/ { print \$2 \$3 \$4 \$5; }"
makefiles/tools/usb_board_reset.mk:  STTY_FLAG = -F
```

So I consider this issue fixed. Please re-open if you disagree.

closed 2020-07-01 by miri64