RIOT-OS/RIOT #8590: Add Flag in Makefile for USBasp

Citrullin, 2018-02-18:
Hi,
I have found any board or flag for using RIOT with an ISP. I got some boards which I can only flash via an ISP. It would be also nice to be able to flash the MCU, without a board. I would suggest to use a new flag for this, since the board flag also includes already the MCU. 

Maybe another solution would be also suitable :)

gebart, 2018-02-19:
My preferred solution here would be to use the DEBUG_ADAPTER environment variable in the same way as for the openocd compatible boards. Setting DEBUG_ADAPTER=usbasp should use the usbasp debugger hardware, and DEBUG_ADAPTER=something else would use the onboard programmer.

miri64, 2018-10-17:
@Citrullin @gebart can this issue be closed?

gebart, 2018-10-17:
I guess this is still unresolved, AFAIK there is no usbasp support yet

closed 2019-09-10 by stale[bot]