RIOT-OS/RIOT #8591: CLion + RIOT (Maybe plaftormio integration) ?

Citrullin, 2018-02-18:
Since RIOT uses a Makefile and CLion supports only CMake, I couldn't find any solution to getting CLion and RIOT work together. Maybe I try now Eclipse, since there is a how to for it. :)

I used platformio before. They have a nice, simple to use, solution. Maybe it makes sense to add RIOT as Library to platformio. This would make it possible to use RIOT with every IDEs in just a few commands. Platformio supports all common used IDEs. (http://docs.platformio.org/en/latest/ide.html)

gebart, 2018-02-19:
What is CLion and what is the benefit of using it in RIOT? What is platformio?

Citrullin, 2018-02-19:
Hi @gebart,
CLion is an IDE by Jetbrains. (Well known for they IntelliJ IDE): https://www.jetbrains.com/clion/
Platformio is a kind of build tool-set: http://docs.platformio.org/en/latest/what-is-platformio.html
They have a CLI for initializing project. So you just run:
```
platformio init --ide clion --board [BOARD_NAME]
```
and they create all necessary stuff for your IDE. You get also a platformio.ini file, which described how your project will flash. In my case: I use the USBasp. It includes also the dependencies, but you can still add some by yourself.

basilfx, 2018-02-23:
I haven't looked into the details, but it seems that there are already many supported frameworks: https://platformio.org/frameworks/arduino. RIOT could be one of them IMHO.

douardda, 2018-11-06:
I do use platformio with mbed on nucleo boards (and also for arduino stuff) and it would be really nice to be able to try riot as easily as:
```
pio init -b nucleo_f446re -O framework=riot
```

Would be awesome!

aabadie, 2018-12-06:
I think this is certainly interesting, other people also told me IRL they were interested.

Anyone willing to have a look ?

andrea-nisti, 2018-12-07:
I'm not that expert since I started embedded just recently, the integration between the two would led to a really nice and moder setup. Something beginners can play with too and professional work with.

I will try at a certain point but I will need lot of guide from the communities.

closed 2019-08-10 by Citrullin