RIOT-OS/RIOT #8596: emb6: dead url to documentation pdf

bergzand, 2018-02-19:
#### Description

The URL in the emb6 docs to the [pdf](https://github.com/hso-esk/emb6/raw/develop/doc/pdf/emb6.pdf)([here](https://github.com/RIOT-OS/RIOT/blob/master/pkg/emb6/doc.txt#L6)) is dead. I don't know what was in the original PDF, so I don't know what the replacement URL should be.

#### Steps to reproduce the issue

Click the url to the pdf in the doxygen: http://riot-os.org/api/group__pkg__emb6.html

#### Expected results

A pdf file describing emb6

#### Actual results

The github 404 message

#### Versions

Firefox 57.0.4 (64-bit)

miri64, 2018-02-19:
It should suffice to use the `PKG_VERSION` of the emb6 pkg instead of `develop`. Currently don't have the means to fix that nowish, so I'm not mad, if anyone else picks up the task ;-)

bergzand, 2018-02-27:
The whole pdf is removed with [this commit](https://github.com/hso-esk/emb6/commit/d1cb0b424fd8d8301a4ed29e50f6fef14364bb61). Maybe their current github wiki is the best equivalent at the moment.

miri64, 2018-02-27:
Mh, but wouldn't it make more sense to have the documentation of the same version the package is referring to?

Lotterleben, 2018-09-19:
How about linking to https://github.com/hso-esk/emb6/blob/f92a4fdfaf1cf8d4bf4ef07130389ebc54b8fd4c/doc/pdf/emb6.pdf , which contains the pdf right before the pdf deletion commit? (If that's acceptable I'll be happy to open the tiny PR for that)

miri64, 2018-09-19:
> How about linking to https://github.com/hso-esk/emb6/blob/f92a4fdfaf1cf8d4bf4ef07130389ebc54b8fd4c/doc/pdf/emb6.pdf , which contains the pdf right before the pdf deletion commit? (If that's acceptable I'll be happy to open the tiny PR for that)

Maybe better refer to the commit used in the `pkg`'s make file (as proposed in previous comments). With GitHub its always best to use commit-based links instead of branch-based links, unless you are really sure what you are doing. @Lotterleben If you want to I'd appreciate if you could provide that PR.

PR #9962 (2018-09-19): emb6: update link to documentation pdf
### Contribution description
Update the link to the emb6 documentation pdf to its state at the time of PKG_VERSION, since the pdf has since been deleted from https://github.com/hso-esk/emb6/tree/develop/doc/.

### Testing procedure
Copy & paste the new link in doc.txt, find working pdf

### Issues/PRs references
Resolves #8596.
closed 2018-09-19 by bergzand