RIOT-OS/RIOT #8598: CCN Lite: Add "no 8-Bit support" to documentation

Citrullin, 2018-02-19:
I tried one CCN Lite example. While compiling I got many errors. I fixed some of them, until I reached the following point:
```
ccnl-pkt-ccntlv.c:508:5: error: left shift count >= width of type [-Werror]
```
ccntlv line 508
```
    if ( (unsigned int)len >= ((1 << 16) - sizeof(struct ccnx_tlvhdr_ccnx2015_s)))
        return -1;
```

Since CCN Lite shifts 16 bits, I guess 8-Bit MCUs are not supported. I recommend to add this somewhere to the documentation. :) There are also other places where CCN Lite shifts 16 bits.

gebart, 2018-02-19:
I guess ccn lite assumes int is 32 bit. This should be fixed upstream, but I guess you should be able to add a cast to uint32_t on the left shift to fix it.

mfrey, 2018-02-20:
CCN-lite lacks precision in the length of types. My assumption is that this is partially due to the fact that it supports multiple message formats. We've already discussed this briefly and it is already part of the [todo list](https://github.com/cn-uofbasel/ccn-lite/wiki/TODO)

Citrullin, 2018-02-20:
@mfrey Of course, it would be nice to get a fix for it :) But for now, I think a short notice in the documentation is just fine. So, user know that 8-Bit MCUs are not supported at the moment. (just for networking with CCN Lite).

PeterKietzmann, 2019-02-21:
It's documented [here](https://github.com/cn-uofbasel/ccn-lite/wiki/Frequently-asked-questions) in the ccn-lite FAQ. @Citrullin, @cgundogan would you provide that information in the RIOT doc? I'm adding the Hack'n'ACK label for that.

miri64, 2019-08-28:
Ping @Citrullin @cgundogan?

Citrullin, 2019-08-28:
@miri64 I don't really care about this.

cgundogan, 2019-08-28:
see https://github.com/cn-uofbasel/ccn-lite/pull/363. Once merged, this issue should resolve.

miri64, 2019-08-28:
Merged, so I guess all we need now is a version bump of CCN-lite to close this issue ;-).

mfrey, 2019-08-28:
> Merged, so I guess all we need now is a version bump of CCN-lite to close this issue ;-).

Afair, last time this broke a few things with the shell commands. But I might be wrong.

miri64, 2019-08-28:
> Afair, last time this broke a few things with the shell commands. But I might be wrong.

For this we have testing.

cgundogan, 2019-08-29:
@Citrullin just out of curiosity: which AVR board did you use to compile ccn-lite?

closed 2019-08-30 by miri64

Citrullin, 2019-08-30:
> @Citrullin just out of curiosity: which AVR board did you use to compile ccn-lite?

Some Arduino clone as I remember. Some ATMEGA328P.

PR #12124 (2019-08-30): pkg: ccn-lite: version bump
### Contribution description
Bumps the ccn-lite vesion.

### Testing procedure
1. A simple Interest-Data exchange test with `native` should suffice.
2. The ccn-lite example should compile for `arduino-uno`(after adding it to the `Makefile`'s whitelist for boards. Note that linking will fail due to the memory limits of the uno, though.

### Issues/PRs references
 Fixes #8598 .