RIOT-OS/RIOT #8631: at86rf2xx/kw2xrf: scalar NETOPT options checked as arrays

miri64, 2018-02-23:
For the `get` and `set` options of both `at86rf2xx` and `kw2xrf` (and maybe more devices, but I did not care to look), the width of some scalar values is assumed to be like an array (the check is not done with a `!=` operator but with `<`, `>`, `<=` or `>=`) but than set as a scalar value by way of pointer dereferencing. I noticed it first for `NETOPT_MAX_PACKET_SIZE`, but I suspect there are more:

```C
        case NETOPT_MAX_PACKET_SIZE:
            if (len < sizeof(int16_t)) {
                return -EOVERFLOW;
            }

            *((uint16_t *)value) = KW2XRF_MAX_PKT_LENGTH - _MAX_MHR_OVERHEAD;
            return sizeof(uint16_t);
```

While for unsigned numbers on little-endian platforms this shouldn't be a problem, signed numbers or on big endian platform the user might get returned a non-error, but the value might not be as expected. E.g. if in the above example `len == sizeof(uint32)` on the setting of a value results in a memory region like this:

```
 0                   1                   2                   3
 0 1 2 3 4 5 6 7 8 9 0 1 2 3 4 5 6 7 8 9 0 1 2 3 4 5 6 7 8 9 0 1
+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
|----------value_space----------|0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0|
+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+
```

On a little-endian platform uint32_t be converted correctly, since they are in the most significant place, on big-endian platforms they would however be in the most significant place, making the number really large. The other way around it would be for the `set` case as on big-endian platforms the 16 most significant bits would be taken, which are 0 here.

To fix this constructs like the one check should be transferred to a

```C
if (len != sizeof(uint16_t)) {
    return -EOVERFLOW;
}
```

or `assert(len == sizeof(uint16_t));` (which would be preferred anyway).

A subset of this issue was already solved in https://github.com/RIOT-OS/RIOT/pull/7488, but there are more problems.

gebart, 2018-02-23:
I think the root cause of this construct is that the documentation for some of the original netopt options still do not contain specifications on the size and data type of the value. All new netopt options have the expected data type specified in the doxygen in the netopt header file (much thanks to your detailed reviews @miri64 :) ), but not the older ones.

miri64, 2018-02-23:
> I think the root cause of this construct is that the documentation for some of the original netopt options still do not contain specifications on the size and data type of the value. All new netopt [...]

At least the ones I and @haukepetersen introduced in the beginning contain this specification pretty much from day 0 (compare c5c9c3dfab89bda864064eb39648a6cc42a2da22). ;-)

gebart, 2018-02-23:
You are right, the majority of the options have data type specified in the docs. We do need to improve the docs for some others though.

miri64, 2018-02-23:
> We do need to improve the docs for some others though.

Yes. And we need to fix this issue ;-)

gebart, 2018-02-23:
I agree! 👍

gebart, 2018-02-27:
I am working on a documentation improvement for netopt.h

gebart, 2018-02-28:
See #8655 

That PR only fixes documentation, no implementations.

miri64, 2019-08-10:
I think the clarification of the doc should be enough for now. The implementations follow slowly, but they do.

closed 2019-08-10 by miri64