RIOT-OS/RIOT #8637: Add custom radio, which SPI conf?

Citrullin, 2018-02-25:
I added a CC1101 to my existing Stm32f103c8 board. The documentation says, I need to include the boards default one: https://doc.riot-os.org/group__net__gnrc.html
Since I don't have a default one, I wanted to add the CC110x driver. Which doesn't work, because it misses the params for it. I figured out that maybe I should change board configurations instead of my applications ones. So I added the ``USEMODULE += cc110x`` to the Makerfile.include and also added the params file to the include folder. I hope that is the correct way to do it. If not, please tell me. 

So, now the cc110x params, which are just SPI pin configuration. I looked into the API docu:
https://riot-os.org/api/structcc110x__params.html 
And this says: what does it look like. Maybe obvious for some people but not for me. I have some questions.

CC1101 MOSI == cc110x_params.spi ?
What pin number do I need to insert there? I got the MCU in the LQFP48 size. I read the specifications: http://www.st.com/content/ccc/resource/technical/document/datasheet/33/d4/6f/1d/df/0b/4c/6d/CD00161566.pdf/files/CD00161566.pdf/jcr:content/translations/en.CD00161566.pdf

So, I guess it is: 
MOSI => 41
MISO => 40
CS => 14

And the gdo pins, for testing, to other, not reserved, GPIO pins.

PeterKietzmann, 2018-02-26:
@Citrullin have a look at #8437. Maybe you can review and test it with your board later? As your board does not define a `CC110X_PARAMS_BOARD` the default configuration `CC110X_PARAMS_DEFAULT` will be used. The pins (e.g. `CC110X_PARAM_GDO0) you can set in your (application ?!) Makefile, for example.

Citrullin, 2018-02-26:
@PeterKietzmann 

```
.cs   = GPIO_PIN(1, 21), 
.gdo0 = GPIO_PIN(0, 27), 
.gdo1 = GPIO_PIN(1, 23), 
.gdo2 = GPIO_PIN(0, 28)
```
That makes totally sense to me. :) Thanks. I'll take later a look into it.

PeterKietzmann, 2018-02-26:
This configuration should work, as it is the hard-wired configuration of the msba2 platforms. However, it is just the default for this single board. For all others, if not defines elsewhere, the fallback is [this configuration](https://github.com/RIOT-OS/RIOT/pull/8437/files#diff-f8ae35bd3ea4db5c3ec7fab79ed91374R48). So you should still define it by yourself. Pleas close the issue once clarified.

aabadie, 2018-04-14:
@Citrullin, could you finally solve your problem with the given answer ? Can we close this issue ?

Citrullin, 2018-04-16:
@aabadie Hi, I'm not that good in C. So I had a bit trouble with getting it running. I will check this again.

PeterKietzmann, 2018-05-29:
@Citrullin please refer to #9225, #9230, #9231. I think for the configuration everything should be pretty clear now. If now, don't hesitate to re-open the issue. I'll close this one for now

closed 2018-05-29 by PeterKietzmann