RIOT-OS/RIOT #8658: adxl345.h MCU Pin Definition in module header.

Josar, 2018-02-28:
https://github.com/RIOT-OS/RIOT/blob/2cab4292d57cd6ec5d16752cfc60ee8030ed423d/drivers/include/adxl345.h#L95

This should not be defined here. IMHO.

Collision with iom256rfr2.h

https://github.com/RIOT-OS/RIOT/pull/8643#discussion_r171176311

dylad, 2018-02-28:
Hi @Josar 
I think this declaration is at the perfect place, this enum is here for select the interrupt type output for the ADXL345's interrupts pins as this accelerometer have two pins dedicated to interrup. Thus there are not related to any CPU.
One solution for your issue would be to rename this to : 
```
enum {
ADXL345_INT1,
ADXL345_INT2
}
```
I can open a PR to change it if you want.

Josar, 2018-02-28:
@dylad  OK thanks.

Josar, 2018-02-28:
What si with stream and fifo would it be better to prepend them also to avoid future collision?

dylad, 2018-02-28:
@Josar #8661 should solve your issue.
I've renamed the two enums to avoid any conflict in the future.

dylad, 2018-02-28:
Oh sorry I didn't see you send a PR just before mine. I'll drop mine then.

dylad, 2018-03-02:
Can we close this issue now that #8696 is merged ?

closed 2018-03-02 by Josar