RIOT-OS/RIOT #8662: nucleo-l152: LSE broken

kYc0o, 2018-02-28:
#### Description

On nucleo-l152 rev C-03 or newer, there's a problem using the on-board 32.768KHz crystal. #8545 proposes to use the internal low speed oscillator to avoid the problem, but it doesn't solve it.
#### Steps to reproduce the issue

Compile any example with LSE set at 1 (CLOCK_LSE define) for BOARD=nucleo-l152.
#### Expected results

Any example works properly.
#### Actual results

It hangs.

aabadie, 2018-02-28:
To be complete, #8545 changes should be updated with a reference on this issue (see [here](https://github.com/RIOT-OS/RIOT/pull/8545/files/f35edcacf1663d05fcb494e3da85d053660aa1e0#diff-5d69d844a724995a1e1add8ee4bbca0c))

kaspar030, 2018-04-11:
What'S the state here?

aabadie, 2018-04-16:
Ok, I don't understand why (maybe #8735 ?), but I just switched CLOCK_LSE to 1 in the nucleo-l152 peripheral configuration and it works like a charm.

So I would say that this issue can be closed.

aabadie, 2018-04-16:
This is issue will be closed automatically when #8961 is merged.

PR #8961 (2018-04-16): boards/nucleo-l152: remove reference to solved issue

### Contribution description

This PR remove the reference to #8662 in the CLOCK_LSE configuration comment, since the issue is not present anymore.

### Issues/PRs references

Fixes #8662
closed 2018-04-16 by kYc0o